egguy/grott-home-assistant-add-on #5: Not seeing all sensors and error from HA log

cyberdiamond, 2023-01-25:
I have installed HA os and have been adding data to it.

To start I was manually creating everything from yaml and that was working but I had only got half the sensors done then I saw your add-on.

The first time I added your add-on before the latest merge into grott I did get all sensors populated (from memory it may have taken a few attempts to fully populate) and was having some issues with the values not being correct, mostly wrong (or maybe no) dividers on some of the data coming in?

I then did a grott pull and updated (stopped / rebuild) the add-on which didn't help, so I uninstalled, removed all sensors manually and reinstalled again with a full HA reboot between.

I only get a few sensors listed now even after multiple rebuild and reinstall attempts, they are:

datalogger_serial
pv_input_actual
pv1_current
pv1_voltage
serial
state

The sensors I do have above do get updated properly.

My setup is:

Growatt SPH-5000 with 2x batteries connected and Growatt / Eastron sdm230 meter
Shinelan box set to send data to Grott IP
Grott running stand alone on top of linux vm pulled from git (so version 2.7.8)
Grott configured with inverter type as sph and your HA extension setup sending to HA os IP (also writing to another linux vm for influxdb).
HA os fully up-to-date running as a vm configured with MQTT broker and your Grott - HA extension - current add-ons installed.
Configuration off your add-on all default except inverter type also set to sph, Disable grott MQTT enabled and Watchdog not enabled.

Error message from HA log:

```
Exception in message_received when handling msg on 'homeassistant/grott/INVERTER-SERIAL-NUMBER/state': '{"datalogserial": "DATALOGGER-SERIAL-NUMBER", "pvserial": "INVERTER-SERIAL-NUMBER", "pvstatus": 5, "pvpowerin": 1400, "pv1voltage": 3186, "pv1current": 2, "pv1watt": 906, "pv2voltage": 2525, "pv2current": 2, "pv2watt": 556, "pvpowerout": 703, "pvfrequentie": 4999, "pvgridvoltage": 2320, "pvgridcurrent": 10, "pvgridpower": 707, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime": 161074232, "eactoday": 378, "pvenergytoday": 378, "eactotal": 207145, "epvtotal": 193644, "epv1today": 233, "epv1total": 126733, "epv2today": 187, "epv2total": 98192, "pvtemperature": 302, "pvipmtemperature": 281, "pvboosttemp": 285, "bat_dsp": 534, "pbusvolt": 4279, "eacharge_today": 0, "eacharge_total": 3537, "batterytype": 1, "uwsysworkmode": 5, "systemfaultword0": 0, "systemfaultword1": 0, "systemfaultword2": 0, "systemfaultword3": 0, "systemfaultword4": 0, "systemfaultword5": 0, "systemfaultword6": 0, "systemfaultword7": 0, "pdischarge1": 0, "p1charge1": 0, "vbat": 531, "SOC": 66, "pactouserr": 0, "pactousertot": 0, "pactogridr": 6200, "pactogridtot": 6200, "plocaloadr": 0, "plocaloadtot": 0, "spdspstatus": 5, "spbusvolt": 3022, "etouser_tod": 1, "etouser_tot": 3436, "etogrid_tod": 550, "etogrid_tot": 248263, "edischarge1_tod": 14, "edischarge1_tot": 54658, "eharge1_tod": 17, "eharge1_tot": 45170, "elocalload_tod": 4294967165, "elocalload_tot": 4294925601, "grott_last_push": "2023-01-25T08:59:35.913628+00:00"}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 44, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 305, in message_received _update_state(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 265, in _update_state payload = self._template(msg.payload, PayloadSentinel.DEFAULT) File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 249, in async_render_with_possible_json_value rendered_payload = self._value_template.async_render_with_possible_json_value( File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 589, in async_render_with_possible_json_value return _render_with_context(self.template, compiled, **variables).strip() File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1950, in _render_with_context return template.render(**kwargs) File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 1301, in render self.environment.handle_exception() File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 936, in handle_exception raise rewrite_traceback_stack(source=source) File "<template>", line 1, in top-level template code TypeError: unsupported operand type(s) for /: 'float' and 'NoneType' 

```

Maes152, 2023-02-12:
I have the same problem.
Looks like something is going wrong in this file: [https://raw.githubusercontent.com/egguy/grott/feature/ha-extension/examples/Home%20Assistent/grott_ha.py](url) in the following snippet
`    # Reuse the existing divide value if available and not existing
    # and apply it to the HA config
    layout = conf.recorddict[conf.layout]
    if "value_template" not in payload and key in layout:
        # From grottdata:207, default type is num, also process numx
        if layout[key].get("type", "num") in ("num", "numx") and layout[key].get(
            "divide", "1"
        ):
            payload[
                "value_template"
            ] = "{{{{ value_json.{key} | float / {divide} }}}}".format(
                key=key,
                divide=layout[key].get("divide"),
            )`

Looks like it's not able to determine "divide" and thus returns None which I believe is triggering this error:  /: 'float' and 'NoneType'

egguy, 2023-02-12:
I took a quick look, and good catch. It seems i return a default value for divide, but forgot to do it for the returned value.

Thx for the help.

I will push the new corrected version soon.

egguy, 2023-02-14:
@Maes152 I've updated the container and pushed a new image. It's the 0.0.7-rc3

You should see an addon update in HA. On the first packet of the growatt you should see:
```
 - Grott extension processing started :  grott_ha
	Grott HA 0.0.7-rc3 - creating XXXXXX config in HA
```

Could you try to see if it has solved your problem ?

Maes152, 2023-02-15:
Thanks @egguy. I feel like an idiot but how am I supposed to update? I am running your "-current" addon and it is not showing any updates available. There is also no rebuild option.. ;(

Maes152, 2023-02-15:
Nvm. Uninstall/install did the job.

Maes152, 2023-02-15:
Hi @egguy, problem is definitely solved.
I think I have a new problem now though and it's scaring me a bit. 
The "Generated PV energy (total) that I am now getting from your addon is currently showing me a number of 8.753,1. But the sensor I added myself 2 days ago is still showing 9.146,9. That's a 393Kwh difference (ouch). 
The official growatt website is also showing me 9146,9 so I have to assume that this is the correct number..

Trying to find out what's going on now.

egguy, 2023-02-15:
Thx for the bug reports

egguy, 2023-02-15:
@cyberdiamond Is it ok for you ?

cyberdiamond, 2023-02-17:
No idea, when I didn't get a reply I just finished off my own mqtt sensors config, which allowed me to configure it all correctly and get all the correct values into HA.

I should be able to test it against a new HA vm and still publish my own mqtt to my original HA vm, so I will try that.

magic-k, 2023-03-01:
I have the same problem, seeing only the above values. But with a different error message:

2023-03-01 09:07:06.668 ERROR (MainThread) [homeassistant.util.logging] Exception in message_received when handling msg on 'homeassistant/grott/JLE4xxxxx/state': '{"datalogserial": "JPCxxxxx", "pvserial": "JLExxxxx", "pvstatus": 1, "pvpowerin": 4695, "pv1voltage": 2418, "pv1current": 9, "pv1watt": 2176, "pv2voltage": 2799, "pv2current": 9, "pv2watt": 2519, "pvpowerout": 4273, "pvfrequentie": 4998, "pvgridvoltage": 4079, "pvgridcurrent": 9, "pvgridpower": 2115, "pvgridvoltage2": 4097, "pvgridcurrent2": 9, "pvgridpower2": 2124, "pvgridvoltage3": 4141, "pvgridcurrent3": 8, "pvgridpower3": 1908, "pvenergytoday": 1, "pvenergytotal": 104067, "totworktime": 47803887, "pvtemperature": 244, "pvipmtemperature": 285, "epv1today": 0, "epv1total": 50024, "epv2today": 0, "epv2total": 57810, "epvtotal": 107834, "grott_last_push": "2023-03-01T08:07:06.554052+00:00"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 44, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 307, in message_received
    _update_state(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 267, in _update_state
    payload = self._template(msg.payload, PayloadSentinel.DEFAULT)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 249, in async_render_with_possible_json_value
    rendered_payload = self._value_template.async_render_with_possible_json_value(
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 624, in async_render_with_possible_json_value
    return _render_with_context(self.template, compiled, **variables).strip()
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1984, in _render_with_context
    return template.render(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
TypeError: unsupported operand type(s) for /: 'float' and 'NoneType'

magic-k, 2023-03-01:
Just to add, i'm using the grott_ha addon in docker which is currently on 0.6

magic-k, 2023-03-01:
The IP wasn't the datalogger but the ha supervisor. I somehow must have reconfigured the datalogger while debugging. So now i can say that i get all expected values as entities with the current version of the grott addon. Thanks.

closed 2023-03-07 by egguy