egguy/grott-home-assistant-add-on #6: fails when trying to make it local

chirs-51, 2023-02-06:
config 
gmode: proxy
gnomqtt: true
ginvtype: default
ginvtypemap: ""
mqtt: {}

log 

Grott proxy mode started
	 **********************************************************************************
	 - Grott - libscrc not installed, no CRC checking only record validation on length!
	 **********************************************************************************
    self.server.bind((conf.grottip, conf.grottport))
TypeError: 'str' object cannot be interpreted as an integer
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

i think maybe i need libscrc installed but how? :)

egguy, 2023-02-06:
Hello,
libscrc isn't required at all in proxy mode. 

Do you never receive any packet? Or the add-on stop immediately?

chirs-51, 2023-02-07:
sorry my settings in the first massagede is wrong trying to make the local sever run so i can send recuest from home assistant to automate it :) 

its stops emedely but its sems like its gets a packet :/ just tried to run the new update and its the same 

if using defult setting and its start and runs but with the massagede 
	 **********************************************************************************
	 - Grott - libscrc not installed, no CRC checking only record validation on length!
	
all i can copy from log when i try to start with local ip 

settings 
gmode: proxy
gnomqtt: true
ginvtype: default
ginvtypemap: ""
mqtt: {}
ggrottport: 1883
ggrottip: 192.168.1.192

{'decrypt': {'value': 'True'}, 'datalogserial': {'value': 16, 'length': 10, 'type': 'text', 'incl': 'yes'}, 'pvserial': {'value': 76, 'length': 10, 'type': 'text'}, 'date': {'value': 136, 'divide': 10}, 'recortype1': {'value': 150, 'length': 2, 'type': 'num', 'incl': 'no'}, 'recortype2': {'value': 154, 'length': 2, 'type': 'num', 'incl': 'no'}, 'pvstatus': {'value': 158, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 162, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 174, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 178, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 186, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 190, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 194, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 298, 'length': 4, 'type': 'numx', 'divide': 10}, 'pvfrequentie': {'value': 306, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 310, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 314, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 318, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 326, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 330, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 334, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 342, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 346, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 350, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 386, 'length': 4, 'type': 'num', 'divide': 7200}, 'eactoday': {'value': 370, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytoday': {'value': 370, 'length': 4, 'type': 'num', 'divide': 10}, 'eactotal': {'value': 378, 'length': 4, 'type': 'num', 'divide': 10}, 'epvtotal': {'value': 522, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1today': {'value': 394, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 402, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2today': {'value': 410, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2total': {'value': 418, 'length': 4, 'type': 'num', 'divide': 10}, 'pvtemperature': {'value': 530, 'length': 2, 'type': 'num', 'divide': 10}, 'pvipmtemperature': {'value': 534, 'length': 2, 'type': 'num', 'divide': 10}, 'pvboosttemp': {'value': 538, 'length': 2, 'type': 'num', 'divide': 10}, 'bat_dsp': {'value': 546, 'length': 2, 'type': 'num', 'divide': 10}, 'pbusvolt': {'value': 550, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, '#nbusvolt': {'value': 554, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, '#ipf': {'value': 558, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, '#realoppercent': {'value': 562, 'length': 2, 'type': 'num', 'divide': 100, 'incl': 'no'}, '#opfullwatt': {'value': 566, 'length': 4, 'type': 'num', 'divide': 10, 'incl': 'no'}, '#deratingmode': {'value': 574, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'eacharge_today': {'value': 606, 'length': 4, 'type': 'num', 'divide': 10}, 'eacharge_total': {'value': 614, 'length': 4, 'type': 'num', 'divide': 10}, 'batterytype': {'value': 634, 'length': 2, 'type': 'num', 'divide': 1}, 'uwsysworkmode': {'value': 666, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword0': {'value': 670, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword1': {'value': 674, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword2': {'value': 678, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword3': {'value': 682, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword4': {'value': 686, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword5': {'value': 690, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword6': {'value': 694, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword7': {'value': 698, 'length': 2, 'type': 'num', 'divide': 1}, 'pdischarge1': {'value': 702, 'length': 4, 'type': 'num', 'divide': 10}, 'p1charge1': {'value': 710, 'length': 4, 'type': 'num', 'divide': 10}, 'vbat': {'value': 718, 'length': 2, 'type': 'num', 'divide': 10}, 'SOC': {'value': 722, 'length': 2, 'type': 'num', 'divide': 100}, 'pactouserr': {'value': 726, 'length': 4, 'type': 'num', 'divide': 10}, '#pactousers': {'value': 734, 'length': 4, 'type': 'num', 'divide': 10, 'incl': 'no'}, '#pactousert': {'value': 742, 'length': 4, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'pactousertot': {'value': 750, 'length': 4, 'type': 'num', 'divide': 10}, 'pactogridr': {'value': 758, 'length': 4, 'type': 'num', 'divide': 10}, '#pactogrids': {'value': 766, 'length': 4, 'type': 'num', 'd [... cut ...]
    self.server.bind((conf.grottip, conf.grottport))
TypeError: 'str' object cannot be interpreted as an integer
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

egguy, 2023-02-15:
It's weird, that's there's value in the grottip and port. Did you change it in the addon configuration?

For the addon, you don't need to specify anything. It will bind by itself to 0.0.0.0.

The only recommended configurations to change are either the `inverter type` or the `inverter mapping` settings.

If you only have one inverter like a SPH5000 you should configure the `inverter type` to sph for example. If you have multiple, you need to use the `inverter mapping` settings and keep the `inverter` type to default.

Also use the latest version of the edge addon for the moment.

chirs-51, 2023-02-25:
yes because i want to run it in local mode to make home assistant to control my inverter and not the growatt severe 
maybe i'm doing it wrong?

egguy, 2023-02-26:
This version of the addon is grott, it only contain the proxy/fowarder to the grott server.

It receives the packet localy for the growatt server, analyze it and send a message to MQTT/PVOut/InfluxDB... then forward to growatt.

There is a function to block commands from growatt and prevent them controlling the inverter. 

If you want to control it locally, it's grottserver, but it's not packaged in this version of the addon and the interface is a bit rough (direct http call with register values and command values).

As for the ip and port,I don't recommend touching it at all if you want the add-on to be able to process the packets. Theses params are to let the server listen to the local IP of the container, it has no relation with groatt server.

chirs-51, 2023-02-26:
Thanks fir the answer it's a shame yes I know it's a bitt rough but it's seems to be the the only way to make lokal control of the inveter right now. But keep up the good work and I hope a solution soon will be have a great day! 👍