egguy/grott-home-assistant-add-on #8: Am I supposed to use grott or grott-current? ;)

Maes152, 2023-02-12:


Maes152, 2023-02-12:
I'm running the grott-current addon (is that the right one?)
A device was created but it does not include the only entity I need "pvenergytotal".

[image]

I can see that all entities are being published through the MQTT state topic though
`{"datalogserial": "xxxx", "pvserial": "xxxx", "recortype1": 0, "recortype2": 124, "pvstatus": 1, "pvpowerin": 8778, "pv1voltage": 4135, "pv1current": 11, "pv1watt": 4548, "pv2voltage": 3022, "pv2current": 14, "pv2watt": 4230, "pv3voltage": 0, "pv3current": 0, "pv3watt": 0, "pvpowerout": 8762, "pvfrequentie": 5000, "pvgridvoltage": 2303, "pvgridcurrent": 14, "pvgridpower": 3224, "pvgridvoltage2": 2347, "pvgridcurrent2": 14, "pvgridpower2": 3285, "pvgridvoltage3": 2315, "pvgridcurrent3": 14, "pvgridpower3": 3241, "Vac_RS": 4008, "Vac_ST": 4055, "Vac_TR": 3997, "pvenergytoday": 11, "pvenergytotal": 90730, "totworktime": 32882718, "epv1today": 4, "epv1total": 47915, "epv2today": 5, "epv2total": 38909, "epvtotal": 86824, "pvtemperature": 528, "pvipmtemperature": 266, "pvboottemperature": 253, "temp4": 0, "uwBatVolt_DSP": 0, "pbusvolt": 3113, "nbusvolt": 3120, "grott_last_push": "2023-02-12T10:44:06.610015+00:00"}`

Maes152, 2023-02-12:
Fixed this by manually publishing the following topic in MQTT
{"name": "xxxx PV Energy Total", "unique_id": "sensor.maes_total_lifetime_energy_output", "state_topic": "homeassistant/grott/xxx/state", "device": {"identifiers": ["xxx"], "name": "xxx", "manufacturer": "GrowWatt"}, "icon": "mdi:lightning-bolt", "value_template": "{{ value_json.pvenergytotal | int / 10 }}"}

Maes152, 2023-02-12:
Looks like same issue as [https://github.com/egguy/grott-home-assistant-add-on/issues/5](url)

egguy, 2023-02-14:
New version of the addon on edge. It's the RC4

The debug output of the addon when you receive the first packet should look like this
```
	Grott HA 0.0.7-rc4 - creating XXXXXXX config in HA, 69 to push
	Pushing 70 configurations payload to HA
	Configurations pushed
```

I have updated the QoS settings of the publisher and it seems to work a lot better. I've tested on a install with 69 values, but only 30 found, now the 69 are all present.

rcvaneck, 2023-02-15:
I ran into the same problem. This update has fixed it. Now all the sensors are automatically discovered. Fantastic!

Maes152, 2023-02-15:
Confirmed fixed!

closed 2023-02-15 by Maes152

egguy, 2023-02-15:
Thx for confirming, I will do a merge request.