egguy/grott-home-assistant-add-on #9: Any way to use this if you're not running HASS Supervised or HASS OS, i.e. HASS in Docker?

Freekers, 2023-02-14:
To my knowledge, [it's only possible to run add-ons if you're running HASS Supervised or HASS OS](https://www.home-assistant.io/addons/). Is there any way to make use of this add-on if you're running HASS in Docker as well?

Thanks!

Freekers, 2023-02-15:
Thanks for your reply.

Your Docker image seems to be working :) The sensors were automatically created in HASS and Data is flowing in. However, I do see the following warnings/errors in the logging that might need fixing:

```
	 - MQTT jsonmsg: 
			 {"device": "XXXXXXXXXXX", "time": "2023-02-15T20:43:33", "buffered": "no",
			 "values": {"datalogserial": "XXXXXXXXXX", "pvserial": "XXXXXXXXXX",
			 "pvstatus": 0, "pvpowerin": 0, "pv1voltage": 647, "pv1current": 0,
			 "pv1watt": 0, "pv2voltage": 650, "pv2current": 0, "pv2watt": 0,
			 "pvpowerout": 0, "pvfrequentie": 4999, "pvgridvoltage": 2362,
			 "pvgridcurrent": 0, "pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2":
			 0, "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0,
			 "pvgridpower3": 0, "totworktime": 44808891, "pvenergytoday": 85,
			 "pvenergytotal": 46290, "epvtotal": 47779, "epv1today": 43, "epv1total":
			 18890, "epv2today": 44, "epv2total": 28889, "pvtemperature": 243,
			 "pvipmtemperature": 253}}
	 - Grott MQTT topic used : energy/growatt
	 - MQTT send failed: [Errno 99] Cannot assign requested address
```

And:

```
	 **********************************************************************************
	 - Grott - libscrc not installed, no CRC checking only record validation on length!
	 **********************************************************************************
```

Thanks!

egguy, 2023-02-16:
For the libscrc it's normal. The CRC is not used for the proxy mode, there's still sanity check in place. It's mostly used in the case of grottserver.

As for the `send failed`, you can disable the MQTT output in the grott.ini file. e.g.

```ini
[MQTT]
nomqtt = True
```
The HA addon use the data of the MQTT payload, but has it own way of sending (for HA config/autoconfig purpose). You can still have the normal MQTT output (or a second MQTT output to a third party)

Freekers, 2023-02-16:
I managed to get it to work using environment variables in the docker-compose file instead of using the grott.ini config file.
My compose file now looks as follows:

```
services:
    grott:
        image: egguy/grott:latest
        container_name: grott
        restart: unless-stopped
        ports:
            - "5279:5279"
        environment:
            gverbose: False
            gnomqtt: True
            gextension: True
            gextname: grott_ha
            gextvar: '{"ha_mqtt_host": "YOUR_IP", "ha_mqtt_port": "1883", "ha_mqtt_user": "XXXXXX", "ha_mqtt_password": "XXXXXX"}'
```

I have 2 more questions/remarks if you don't mind :)

- Your Dockerfile installs influxdb twice. I assume the second time you meant influxdb-client? But I was wondering; are both really needed for this extension to work? Maybe they can be removed to reduce the image size.
- Instead of using python:3.9 as base image, I propose to use python:3.9-slim as base-image to further reduce the image size. The official Grott docker image is also built using the slim tag.

egguy, 2023-02-16:
For the influxdb dependency, I've imported from the original.
- Influxdb: Connect to InfluxDB 1.X series
- influxdb-client: Connect to InfluxDB 2.X series
The dependencies are lazy loaded, so it could be removed. This image is a generic image for grott and people could use it fto push data in influxDB.
From a look on pypi it weights less than 1MO for the sum of both lib.

For the base image, you are right, it works on 3.9-slim. This is a remain of the multi-archi grott/grottserver build I've tested.
The grottserver require libscrc and to make an arm image you need to compile libscrc which only works on the non slim version of the image (I did some test, and you can perfectly replace by a python CRC implementation to remove this requirement).

If I start to generate CI build of image, I will probably consider this.

Freekers, 2023-02-16:
Thanks for the elaborate answer :)

I will go ahead and close this issue, as my questions have been answered. For those of you who stumble upon this issue, the docker-compose file in my post above should get you going.

Many thanks to @egguy for creating this image, this was was exactly what I needed after Growatt started to ban API users.

closed 2023-02-16 by Freekers