egguy/grott-home-assistant-add-on #11: no updates on shinephone shows offline

bschofield425, 2023-02-16:
I'm use the Proxy mode. 
so I'm able to get some data, many errors with it's implementation into the energy panel, will need to customize the raw data I think ( or wait for some future updates) 
I utilize the energy panel for logic control for other functions, so without it working correctly it definitely puts me at a disadvantage. 
but the biggest problem I've got right now is that I can't compare data with the shinephone app as it shows offline, and I get these errors: 

 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.33.4', 5279), raddr=('192.168.9.248', 51266)>
	 - Data less then minimum record length, data not processed

I'm not sure if this is related with the proxy not working correctly to show me values in the app. 

Thank you

Barry

Freekers, 2023-02-16:
This is normal behavior. I think those are keep-alive packets, I also have many of those in my logging.

Data packets are sent every 5 minutes (by default) from the inverter to Growatt. If a data packet is intercepted, it will show up in the logging as follows (verbose): 

```
	 - Growatt packet received:
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.22.0.24', 45410), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x00\x02\x00\x06\x02\x41\x01\x04\x1f\x35\x2b\x43\x20\x40\x4d\x7e\x38\x28\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x3b\x32\x26\x46\x36\x77\x40\x5f\x44\x32\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x76\x7b\x53\x59\x4e\x75\x6a
		 \xcc\x78\x73\x72\x6f\x77\x61\x74\x74\x45\xf5\x6f\x77\x61\x74\x74\x47\x70\xe5
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x7c\xf0\x68\x4e
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x4e\x48\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x63\xdf\xce\xfc\x72\x6f\x77\x34\x74\x74\xf3\xa0\x6f\x77\xdb\xd7\x74
		 \x47\x72\x44\x77\x61\x3d\xbe\x47\x72\x6f\x5b\x61\x74\x04\x9e\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\xb8\x88\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x87\x47\x81\x6f\x77\x61\x74\x74\xba\x73\xdc\x77\x61\x3a\x54\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x75\x74\x46\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x73
		 \x03\x79\x71\x53\x64\x47\x7b\x6f\x77\x61\x74\x74\x47\x57\x22
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T060104X
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNX
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 000200060241010458474434413439394a4700000000000000000000000000000000000000005
		 4454732423032303353000000000000000000000000000000000000000017020f142b21020bb8
		 0c340000000000000287000000000000028a00000000000000000000000000000000000000000
		 0000000000000000000000000001387093a000000000000000000000000000000000000000000
		 00093a0000000000000000000000000000000002abbabb000000550000b4d20000baa30000002
		 b000049ca0000002c000070d90000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000fffa0000000000000000000000f300f30
		 000000000fd01b300004e20000000000000000000000000000000000000000000000000000000
		 00003c0000000000000000000000000000000000000c350cb1000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000002710000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 000000000000000000000000000100010000000000000000000000000000016c0e10271000090
		 00000000000254d
	 - Possible Inverter serial TEG2B0XXXX
	 - Inverter serial not recognised - using inverter type default
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNX
	 - Grott data record date/time processing started
[... 27 lines cut ...]
		 -  epv1today            :  4.3
		 -  epv1total            :  1889.0
		 -  epv2today            :  4.4
		 -  epv2total            :  2888.9
		 -  pvtemperature        :  24.3
		 -  pvipmtemperature     :  25.3
	 - MQTT jsonmsg: 
			 {"device": "TEG2B02XXXX", "time": "2023-02-15T20:43:33", "buffered": "no",
			 "values": {"datalogserial": "XGD4A4XXX", "pvserial": "TEG2B0XXXX",
			 "pvstatus": 0, "pvpowerin": 0, "pv1voltage": 647, "pv1current": 0,
			 "pv1watt": 0, "pv2voltage": 650, "pv2current": 0, "pv2watt": 0,
			 "pvpowerout": 0, "pvfrequentie": 4999, "pvgridvoltage": 2362,
			 "pvgridcurrent": 0, "pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2":
			 0, "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0,
			 "pvgridpower3": 0, "totworktime": 44808891, "pvenergytoday": 85,
			 "pvenergytotal": 46290, "epvtotal": 47779, "epv1today": 43, "epv1total":
			 18890, "epv2today": 44, "epv2total": 28889, "pvtemperature": 243,
			 "pvipmtemperature": 253}}
	 - Grott Send data to PVOutput disabled 
	 - Grott Send data to Influx disabled 
	 - Grott extension processing started :  grott_ha
	Grott HA 0.0.7-rc4 - creating TEG2B0XXX config in HA, 32 to push
	Pushing 33 configurations payload to HA
	Configurations pushed
````

bschofield425, 2023-02-16:
ok. I'm getting that output, but still no updates on the shinephone app, shows offline. 
is there anything else in the proxy that needs to change ?

Barry

Freekers, 2023-02-16:
For me the inverter also shows offline in ShinePhone if there is no sun. I'm not sure where you are located, but if it's already evening at your place, I suggest to check again tomorrow.

bschofield425, 2023-02-16:
I have sun, in AZ. but it always showed online, with battery info, grid info, load % etc. 
now I'm not getting any data at all passing into it.

Freekers, 2023-02-16:
Ah, I see. I'm no expert in this, but what does it say in the logging for "_Growatt server" just after you enable the add-on?
For example, for me it says:

```
_Growatt server:
	growattip:           	 47.91.67.66
	growattport:         	 5279
```

This means the add-on was able to resolve the Growatt server. Your IP might be different as you're in the US and Growatt has multiple servers.

bschofield425, 2023-02-16:
yes that is the ip address it shows. interesting, I know when i log into the web interface I need to log into server-us.growatt.com. 
any way I can put that dns entry in? 
I will see if I can resolve what this ip is. if so, where would I enter that?

bschofield425, 2023-02-16:
yeah when I ping server-us.growatt.com i get 47.251.35.20

Freekers, 2023-02-16:
Hmm, I wish I knew more troubleshooting steps but sadly I don't. @egguy might be able to help :)

bschofield425, 2023-02-16:
by putting the IP in the configuration ggrowattip it has now redirected to the correct server and I see data on the shinephone app.

closed 2023-02-16 by bschofield425

egguy, 2023-02-16:
You can also use a DNS in the configuration. This is automagically resolved and this should save you from having to reconfigure the software when the IP change.

bschofield425, 2023-02-16:
right in that same field? I wasn't sure whether it would resolve. 
Thank you.

egguy, 2023-02-16:
Yep, it's confusing this is still called grottip, but the python connect() call does the resolving.