egguy/grott-home-assistant-add-on #13: Sensors for energy panel incorrect

bschofield425, 2023-02-17:
due to problems with the original growatt connection I as well as so many others have had to look for solutions, this is working, I am passing data back to shinephone and collecting data, however it looks that there may be a mapping issue as I'm not getting accurate data in some of the sensors, and many of the needed battery and incoming from grid sensors are not available in the energy dashboard. 
let me know if there is something I need to change, assist with, wait for? 

Thank you. 
Barry

egguy, 2023-02-23:
Hello,

Thank you for your patience, i've discovered that github don't notify on forked repository.

For the battery, if you have an SPH or SPF, the problem can come from the default layout. You can change the inverter type in the extension configuration. 

There are also some measures for which I don't have a clear mapping (missing name of the settings or unit), if it's the can, can you indicate me the name of the measure, with the unit?

Thx

bschofield425, 2023-02-23:
Thank you, I do have spf, what setting should i have in the configuration file? I've tried inverter type spf, didn't seem to make  a difference. 

Thank you again for your work.

egguy, 2023-02-23:
This should be this setting:

[image]

bschofield425, 2023-02-23:
ok. will put in again and take a look. 
I will advise on results.

egguy, 2023-02-23:
You can also use the value default and then use the inverter mapping if you know the serial; of your inverter, e.g.

[image]

bschofield425, 2023-02-23:
I will try that, because there is still something wrong with mapping to mqtt. 
attached is the current output, some I had to change names based on what I knew the values were by looking at shinephone app. 
for instance the setting for Battery charge current is mapped as sensor.jvj1bkw00v_pv2_voltage and is set to type voltage. 

there are just still some serious mapping issues. 

Sensors
Battery charge Current
18.5 V
Battery Voltage
47.66
Charge from Grid
0.0
House Load
300.0 W
AC_InVA
0.0
AC_InWatt
0.0
ACCharCurr
0.0
Acchr_VA
0.0
Acchr_watt
0.0
ACDischarVA
342.0
ACDischarWatt
303.0
BatDischarVA
0.0
BatDischarWatt
0.0
BatterySoc
10.0
BatWatt
-800.0
Buck1 current
23.6 A
Buck1_ntc
20.6
Buck2 current
0.0 A
Buck2_ntc
0.0
Bus_volt
204.7
ConstantPowerOK
1257.0
Datalogger serial
PWH0CGQ0K4
Dcdctemp
18.5
EacCharTotal
138.9
EacDischarToday
0.0
EacDischarTotal
166.6
EbatDischarToday
3.7
EbatDischarTotal
247.0
FaultBit
0.0
FaultValue
0.0
Generated energy (Today)
1.0 kWh
Generated energy (Total)
0.0 kWh
Generated PV energy (Total)
305.1 kWh
Grid frequency
0.0 Hz
Grid_volt
123.0
Grott last data push
4 minutes ago
Inv_Curr
3.1
Inverter temperature
66.7 °F
Invtemp
22.1
IPM temperature
32.0 °F
Line_freq
59.95
Loadpercent
10.9
OP_Curr
2.7
Output active power
289.0 W
Output apparent power
83,968.0 VA
Outputfreq
59.98
Outputvolt
120.0
Phase 1 current
0.0 A
Phase 1 power
474.0 W
Phase 1 voltage
120.0 V
Phase 2 current
204.0 A
Phase 2 power
8,107,402.9 W
Phase 2 voltage
1.0 V
Phase 3 current
600.0 A
Phase 3 power
20.7 W
Phase 3 voltage
120.0 V
PV Energy today (Today)
1.0 kWh
PV Input (Actual)
11,783,372.8 W
PV Output (Actual)
289.0 W
PV1 charge power
1,129.0 W
PV1 charge power
0.0 W
PV1 Voltage
0.0 V
PV1 Watt
0.0 W
PV2 Current
0.0 A
PV2 Voltage
0.0 V
Serial
JVJ1BKW00V
Solar PV1 production (Total)
304.7 kWh
Solar PV2 production
0.0 kWh
Solar PV2 production (Total)
0.0 kWh
State
12.0
WarningBit
0.0
WarningValue
0.0
Working time
0
solar generated watt
879.0 A
Solar panel Voltage
186.7 V
Total solar production for the day
0.6 kWh

egguy, 2023-02-23:
Yes, There are some values currently not mapped.

If there's no unit, or the sensor is the name of the json value, it's a sign, it's not fully mapped. I can see for exemple: `Invtemp`

If you know what they match and their unit, I will gladly accept some help. I can integrate in the new version the changes.

I see some:
- BatDischarVA -> battery discharge Apparent Power (VA)?
- OP_Curr -> Operating current ???

bschofield425, 2023-02-23:
I just updated and restarted the add on, doesn't appear to fix the issues. 
here is the log file:

 no matching record layout found, try generic
	 - Record layout used :  T06NNNNSPF
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00050006015f015050574830434751304b3400000000000000000000000000000000000000004
		 a564a31424b573030560000000000000000000000000000000000000000170217092414030000
		 002c000c07270000000034940000000001180000000007ee000009b0000000000000000012bb0
		 00a080e04d0176b04af176e000000eb00c70043000000000000000000dd000000140019000000
		 000000000000000000000000004e89002d005904e9000000000000000f00000bf000000000000
		 00000000000000000056d00000025000009a6000000000000068200000000083e00000a320000
		 000000000000ffffd580000000000000004900e200000000002a00000cf00000005a008600e40
		 00100000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 000000000000000000000565d
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNSPF
	 - Grott data record date/time processing started
	 - date-time:  2023-02-23T09:36:20
	 - Grott values retrieved:
		 -  datalogserial        :  PWH0CGQ0K4
		 -  pvserial             :  JVJ1BKW00V
		 -  pvstatus             :  12
		 -  vpv1                 :  183.1
		 -  vpv2                 :  0.0
		 -  ppv1                 :  1346.0
		 -  ppv2                 :  0.0
		 -  buck1curr            :  28.0
		 -  buck2curr            :  0.0
		 -  op_watt              :  203.0
		 -  pvpowerout           :  203.0
		 -  op_va                :  63488.0
		 -  acchr_watt           :  0.0
		 -  acchr_VA             :  0.0
		 -  bat_Volt             :  48.0
		 -  batterySoc           :  10
		 -  bus_volt             :  206.2
		 -  grid_volt            :  123.2
		 -  line_freq            :  60.0
		 -  outputvolt           :  119.9
		 -  pvgridvoltage        :  119.9
		 -  outputfreq           :  60.0
		 -  invtemp              :  23.5
		 -  dcdctemp             :  19.9
		 -  loadpercent          :  6.7
		 -  buck1_ntc            :  22.1
		 -  buck2_ntc            :  0.0
		 -  OP_Curr              :  2.0
		 -  Inv_Curr             :  2.5
		 -  AC_InWatt            :  0.0
		 -  AC_InVA              :  0.0
		 -  faultBit             :  0
		 -  warningBit           :  0
		 -  faultValue           :  0
		 -  warningValue         :  0
		 -  constantPowerOK      :  1257
		 -  epvtoday             :  1.5
		 -  pvenergytoday        :  1.5
		 -  epvtotal             :  305.6
		 -  eacCharToday         :  0.0
		 -  eacCharTotal         :  138.9
		 -  ebatDischarToday     :  3.7
		 -  ebatDischarTotal     :  247.0
		 -  eacDischarToday      :  0.0
		 -  eacDischarTotal      :  166.6
		 -  ACCharCurr           :  0.0
		 -  ACDischarWatt        :  211.0
		 -  ACDischarVA          :  261.0
		 -  BatDischarWatt       :  0.0
		 -  BatDischarVA         :  0.0
		 -  BatWatt              :  -1088.0
	 - MQTT jsonmsg: 
			 {"device": "JVJ1BKW00V", "time": "2023-02-23T09:36:20", "buffered": "yes",
			 "values": {"datalogserial": "PWH0CGQ0K4", "pvserial": "JVJ1BKW00V",
			 "pvstatus": 12, "vpv1": 1831, "vpv2": 0, "ppv1": 13460, "ppv2": 0,
			 "buck1curr": 280, "buck2curr": 0, "op_watt": 2030, "pvpowerout": 2030,
			 "op_va": 634880, "acchr_watt": 0, "acchr_VA": 0, "bat_Volt": 4795,
			 "batterySoc": 10, "bus_volt": 2062, "grid_volt": 1232, "line_freq": 5995,
			 "outputvolt": 1199, "pvgridvoltage": 1199, "outputfreq": 5998, "invtemp":
			 235, "dcdctemp": 199, "loadpercent": 67, "buck1_ntc": 221, "buck2_ntc": 0,
			 "OP_Curr": 20, "Inv_Curr": 25, "AC_InWatt": 0, "AC_InVA": 0, "faultBit": 0,
			 "warningBit": 0, "faultValue": 0, "warningValue": 0, "constantPowerOK":
			 1257, "epvtoday": 15, "pvenergytoday": 15, "epvtotal": 3056, "eacCharToday":
			 0, "eacCharTotal": 1389, "ebatDischarToday": 37, "ebatDischarTotal": 2470,
			 "eacDischarToday": 0, "eacDischarTotal": 1666, "ACCharCurr": 0,
			 "ACDischarWatt": 2110, "ACDischarVA": 2610, "BatDischarWatt": 0,
			 "BatDischarVA": 0, "BatWatt": -10880}}

bschofield425, 2023-02-23:
also, some of the sensors, like pvpowerout, op_watt, bat_volt epvtoday, pvenergytoday, etc. are not available in the energy Dashboard to use.
is there a way for me to make changes to these ?

egguy, 2023-02-23:
At the moment, there is no easy way to directly change this in the addon, I'm thinking about implementing a way to load the mapping with a JSON file later.

One option is to change the properties on HA, by using this example: https://community.home-assistant.io/t/adding-units-of-measurement-in-2021-12/384972/6

```yaml
homeassistant:
  customize:
    sensor.INVSERIAL_bat_volt:
      unit_of_measurement: "V"
      device_class: voltage
      state_class: measurement
    sensor.INVSERIAL_pvenergytoday:
      unit_of_measurement: "kWh"
      device_class: energy
      state_class: total_increasing
```

If you can give me some who are broken, I could implement them quickly.
      
For the case of `pvpowerout`, If I'm my understanding is right, this is an instant reading of the power, you need to use an external integration to do this calculation for you, a good explanation is here: https://community.home-assistant.io/t/convert-w-to-kwh/327007/3

The sensor to use for this sensor is `pvenergytotal`, it has the state_class `total_increasing` which is necessary to take into account the production. This is a good primer on the difference, https://developers.home-assistant.io/blog/2021/08/16/state_class_total/ it's possible to override the sensor `pvpowerout` with the state_class `total`.

bschofield425, 2023-02-23:
i will look into it, maybe i need to wait a day for some of these values to settle as I'm getting some wonky negative pv value to start the day and then it seems to go correct, and I'm not finding at all values to use for battery charge and discharge. 
I will let you know what I find. 
thank you again

bschofield425, 2023-02-24:
I've tried making some changes to configuration.yaml, nothing is making these show up in the Energy dashboard, 
they do show correct values now when viewed. 

homeassistant:
    customize:
        sensor.jvj1bkw00v_pv2_voltage:
            unit_of_measurement: "A"
            device_class: current
            state_class: measurement
        sensor.jvj1bkw00v_bat_volt:
            unit_of_measurement: "V"
            device_class: voltage
            state_class: measurement
        sensor.jvj1bkw00v_ac_inwatt:
            unit_of_measurement: "kwh"
            device_class: energy
            state_class: total_increasing
        sensor.jvj1bkw00v_ebatdischartoday:
            unit_of_measurement: "kwh"
            device_class: energy
            state_class: total_increasing

egguy, 2023-02-28:
There's a new version which has been pushed. This version contains multiple rectification of measures.

bschofield425, 2023-02-28:
I just updated, I can now get a proper Solar power status in the energy panel, still not getting proper battery options

here is my log on startup

 layout   :  T060104SPF
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNSPF
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 006e0006015f010450574830434751304b3400000000000000000000000000000000000000004
		 a564a31424b57303056000000000000000000000000000000000000000017021c08362d030000
		 002c000c07100000000028f00000000000de000000000c5800000e2e0000000000000000126b0
		 00a07ec04d0176d04b01771000000d500ae0079000000000000000000c80000001e0022000000
		 000000000000000000000000004e89002d005904ed000000000000000700000e7c00000000000
		 0000000000000000005740000002200000b3400000000000006d4000000000cc600000eb00000
		 000000000000ffffe4ee000000000000003d009200000000002500000f770000005a008600930
		 00100000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 0000000000000000000006950
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNSPF
	 - Grott data record date/time processing started
	 - date-time:  2023-02-28T08:54:45
	 - Grott values retrieved:
		 -  datalogserial        :  PWH0CGQ0K4
		 -  pvserial             :  JVJ1BKW00V
		 -  pvstatus             :  12
		 -  vpv1                 :  180.8
		 -  vpv2                 :  0.0
		 -  ppv1                 :  1048.0
		 -  ppv2                 :  0.0
		 -  buck1curr            :  22.2
		 -  buck2curr            :  0.0
		 -  op_watt              :  316.0
		 -  pvpowerout           :  316.0
		 -  op_va                :  92928.0
		 -  acchr_watt           :  0.0
		 -  acchr_VA             :  0.0
		 -  bat_Volt             :  47.1
		 -  batterySoc           :  10
		 -  bus_volt             :  202.8
		 -  grid_volt            :  123.2
		 -  line_freq            :  60.0
		 -  outputvolt           :  120.0
		 -  pvgridvoltage        :  120.0
		 -  outputfreq           :  60.0
		 -  invtemp              :  21.3
		 -  dcdctemp             :  17.4
		 -  loadpercent          :  12.1
		 -  buck1_ntc            :  20.0
		 -  buck2_ntc            :  0.0
		 -  OP_Curr              :  3.0
		 -  Inv_Curr             :  3.4
		 -  AC_InWatt            :  0.0
		 -  AC_InVA              :  0.0
		 -  faultBit             :  0
		 -  warningBit           :  0
		 -  faultValue           :  0
		 -  warningValue         :  0
		 -  constantPowerOK      :  1261
		 -  epvtoday             :  0.7
		 -  pvenergytoday        :  0.7
		 -  epvtotal             :  370.8
		 -  eacCharToday         :  0.0
		 -  eacCharTotal         :  139.6
		 -  ebatDischarToday     :  3.4
		 -  ebatDischarTotal     :  286.8
		 -  eacDischarToday      :  0.0
		 -  eacDischarTotal      :  174.8
		 -  ACCharCurr           :  0.0
		 -  ACDischarWatt        :  327.0
		 -  ACDischarVA          :  376.0
		 -  BatDischarWatt       :  0.0
		 -  BatDischarVA         :  0.0
		 -  BatWatt              :  -693.0
	 - MQTT jsonmsg: 
			 {"device": "JVJ1BKW00V", "time": "2023-02-28T08:54:45", "buffered": "no",
			 "values": {"datalogserial": "PWH0CGQ0K4", "pvserial": "JVJ1BKW00V",
			 "pvstatus": 12, "vpv1": 1808, "vpv2": 0, "ppv1": 10480, "ppv2": 0,
			 "buck1curr": 222, "buck2curr": 0, "op_watt": 3160, "pvpowerout": 3160,
			 "op_va": 929280, "acchr_watt": 0, "acchr_VA": 0, "bat_Volt": 4715,
			 "batterySoc": 10, "bus_volt": 2028, "grid_volt": 1232, "line_freq": 5997,
			 "outputvolt": 1200, "pvgridvoltage": 1200, "outputfreq": 6001, "invtemp":
			 213, "dcdctemp": 174, "loadpercent": 121, "buck1_ntc": 200, "buck2_ntc": 0,
			 "OP_Curr": 30, "Inv_Curr": 34, "AC_InWatt": 0, "AC_InVA": 0, "faultBit": 0,
			 "warningBit": 0, "faultValue": 0, "warningValue": 0, "constantPowerOK":
			 1261, "epvtoday": 7, "pvenergytoday": 7, "epvtotal": 3708, "eacCharToday":
			 0, "eacCharTotal": 1396, "ebatDischarToday": 34, "ebatDischarTotal": 2868,
			 "eacDischarToday": 0, "eacDischarTotal": 1748, "ACCharCurr": 0,
			 "ACDischarWatt": 3270, "ACDischarVA": 3760, "BatDischarWatt": 0,
			 "BatDischarVA": 0, "BatWatt": -6930}}
	 - No MQTT message sent, MQTT disabled
	 - Grott Send data to PVOutput disabled 
	 - Grott Send data to Influx disabled 
	 - Grott extension processing started :  grott_ha
	 - Grott extension processing ended :  0
	 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.33.4', 5279), raddr=('192.168.9.248', 24872)>
	 - Data less then minimum record length, data not processed

egguy, 2023-02-28:
What is missing with the batteries?

If you look at this issue: https://github.com/egguy/grott-home-assistant-add-on/issues/14#issuecomment-1448229246 you can see What remain to be mapped.

bschofield425, 2023-02-28:
the following 2 values are not showing in the energy dashboard to choose to use. 

[image]

bschofield425, 2023-02-28:
I see it looks like those are still to be mapped, thank you for your work, will wait patiently... this is still significantly better than the problems with the growatt integration being unreliable.

bschofield425, 2023-04-11:
I have been using this add in and am quite happy with all except for the battery functions in the energy dashboard. 
is there any update with this?

Thank you

egguy, 2023-04-14:
Hello,
Can you take a look in `Devloper tools -> States` if you can filter on the battery, It should show something like that:
[image]
If the device_class and state are not like this, that's means there can be a misconfiguration.

If it's the case, can you try to restart the addon ? And in the meantime can I have the name of the sensor so i can take a look in the code?

bschofield425, 2023-04-14:
Thank you for getting back to me, discharge value is showing correct, but it does not show up in the Energy panel as an option, not by the entity name or friendly name

egguy, 2023-04-17:
Can you tell me how the sensor is configurer in the dev panel, or in the MQTT? This would help me to diagnose the problem and publish a bug fix..

egguy, 2023-04-18:
Thank you.

Can you share with me the parameters of the sensor ? Like in this link: https://github.com/egguy/grott-home-assistant-add-on/issues/13#issuecomment-1507757000 This would help me to diagnose why it isn't available in the energy panel.

bschofield425, 2023-04-18:
here is what I have for this, it all looks correct that is what is so frustrating:

[image]

egguy, 2023-05-16:
Hi,
I'm still trying to understand why it's not working.
The preferred class should be total_increasing (discharged total).

I took a deep dive in HA. it seems it's validates all the required params for being displayed. This is weird. I will to look at others params to see if it has an influence.

bschofield425, 2023-09-14:
due to updates, and also clearing all stale entities this has now been resolved

closed 2023-09-14 by bschofield425