egguy/grott-home-assistant-add-on #14: Offgrid SPF 5000 ES - missing values

blackbird0589, 2023-02-21:
Hi egguy,
I've got a Growatt inverter and using your grott add-on on my HA.
For now I've trieded everything but don't get all values of the SPF. It only has one MPP-tracker and a battery connected.
In HA MQTT can be seen pv1 with regular voltages and a pv2 with 0V, but no battery-states (power, energy, SOC). If I activate the option "gincludeall" the values are published, but not rated (54V are displayed as 5400 without unit).
Is there a way to help me?

 thanks in advance

egguy, 2023-02-23:
Hi,

In the configuration of the add-on (the edge is the recommended version ATM), did you change the inverter type from default to sph ? If not enabled, you miss some information. 

ref: https://github.com/johanmeijer/grott/blob/0d1e6878b6d0bd228e4ff3d351fd92c6c389d320/README.md#spf-off-grid-inverter-support

blackbird0589, 2023-02-23:
Hello again,
and thank you - that's one part of the solution! Now I can see every value in the automaticly generated mqtt-device.
But for most parts only numbers without unit.
[image]

Is there a solution for that problem

egguy, 2023-02-23:
It's part of some of the unimplemented mappings, if you can do me a list of them, I will be glad to implement them quickly.

If you can do something like this, this would help me a lot:
- bus_volt: Solar bus voltage | V

Some possible temporary/non-scalable, solutions are listed here: https://github.com/egguy/grott-home-assistant-add-on/issues/13#issuecomment-1442485832

blackbird0589, 2023-02-24:
Ok, I tried do complete this table but some parts are missing because I don't know the exactly meanings of the values.
Because github doesn't support yaml-files i converted it into txt - also see below.
[Growatt SPF5000ES value-table.txt](https://github.com/egguy/grott-home-assistant-add-on/files/10822906/Growatt.SPF5000ES.value-table.txt)

datalogserial: Inverter serial | 
pvserial: Datalogger serial | 
pvstatus: Inverter status | 
vpv1: PV string voltage | V
vpv2: ---- no values, inverter has only one string
ppv1: PV power | W
ppv2: ---- no values, inverter has only one string
buck1curr: PV current to storage | A
buck2curr---- no values, inverter has only one string
op_watt: inverter consumption power | W
pvpowerout: pv power output | W
op_va: inverter reactive power | VA   ### value is not correct
acchr_watt: Storage charging grid | W
acchr_VA: Storage charging grid reactive power | VA
bat_Volt: Battery Voltage | V
batterySoc: Battery state of charge | %
bus_volt: Inverter bus voltage | V
grid_volt: Grid voltage | V
line_freq: Grid line frequency | Hz
outputvolt: Output voltage | V
pvgridvoltage: Inverter output voltage | V
outputfreq: Inverter output freqency | Hz
invtemp: Inverter temperature | °C    ### don't know if unit is possible to display
dcdctemp: Battery charger temperature | °C
loadpercent: Load percentage | %
buck1_ntc:    ### not sure what this value is about | A
buck2_ntc---- no values, inverter has only one string
OP_Curr: Inverter consumption current | A
Inv_Curr: Inverter current | A
AC_InWatt: Grid input power | W
AC_InVA: Grid input reactive power | VA
faultBit: Fault message | 
warningBit: Warning message |
faultValue: Fault value |
warningValue: Warning value |
constantPowerOK: ### not sure what this value is about
epvtoday: PV energy today | kWh
(pvenergytoday: PV energy today | kWh)    #### not sure whats the difference to epvtoday
epvtotal: PV energy total | kWh
eacCharToday: Battery charging grid today | kWh
eacCharTotal: Battery charging grid total | kWh
ebatDischarToday: Battery discharge today | kWh
ebatDischarTotal: Battery discharge total | kWh
eacDischarToday: Grid consumption today | kWh
eacDischarTotal: Grid consumption total | kWh
ACCharCurr: Battery charging current
ACDischarWatt: Load power | kWh
ACDischarVA: Load reactive power | VA
BatDischarWatt: Battery discharge power | W
BatDischarVA: Battery discharge reactive power | VA
BatWatt: Battery discharge power | W  #### not sure whats the difference to BatDischarWatt

egguy, 2023-02-26:
Thank you so much for all the information. I will try to back port this information to the main plugin soon.

blackbird0589, 2023-03-12:
Hi Etienne, I've tried out actual edge-4355431636...your implements works so far very well. Would appreciate if you would have some time to add the open points on the list.

In any case thank you very much for your fast support and your time!

egguy, 2023-03-18:
Hi, thank you for your patience. I'm a bit busy at the moment. 
I've added the last 5 items, for the last 3 I'm waiting until I have further information on it.

closed 2023-09-18 by github-actions[bot]