egguy/grott-home-assistant-add-on #16: Wrong Units & Values - SPH5000 & MIN 5000TL-X

FlyAndy123, 2023-02-28:
Hello,

I installed the HA addon and am using HassIO and have two Growatt inverters: SPH5000 and MIN 5000TL-X
Most of the correct fields seem to be created but there are a few issues with the units being wrong.
For example, battery SOC is reported as 100 = 1%, 50 = 0.5%
[image]

I cant see how I can correct this easily.

I've copied the MQTT JSON message below:

 - MQTT jsonmsg: 
			 {"device": "NTCNAXXXXM", "time": "2023-02-28T11:28:22", "buffered": "no",
			 "values": {"datalogserial": "NAC5AXXXXE", "pvserial": "NTCNAXXXXM",
			 "pvstatus": 5, "pvpowerin": 23500, "pv1voltage": 3654, "pv1current": 355,
			 "pv1watt": 13204, "pv2voltage": 2935, "pv2current": 354, "pv2watt": 10308,
			 "pvpowerout": 0, "pvfrequentie": 5004, "pvgridvoltage": 2435,
			 "pvgridcurrent": 31, "pvgridpower": 0, "pvgridvoltage2": 0,
			 "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0,
			 "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime": 110084478,
			 "eactoday": 50, "pvenergytoday": 50, "eactotal": 189050, "epvtotal": 206270,
			 "epv1today": 53, "epv1total": 118009, "epv2today": 42, "epv2total": 87890,
			 "pvtemperature": 487, "pvipmtemperature": 494, "pvboosttemp": 441,
			 "bat_dsp": 542, "eacharge_today": 10, "eacharge_total": 6957, "batterytype":
			 1, "uwsysworkmode": 5, "systemfaultword0": 0, "systemfaultword1": 0,
			 "systemfaultword2": 0, "systemfaultword3": 0, "systemfaultword4": 0,
			 "systemfaultword5": 0, "systemfaultword6": 0, "systemfaultword7": 0,
			 "pdischarge1": 0, "p1charge1": 29900, "vbat": 538, "SOC": 75, "pactouserr":
			 0, "pactousertot": 0, "pactogridr": 2600, "pactogridtot": 2600,
			 "plocaloadr": 0, "plocaloadtot": 0, "spdspstatus": 5, "spbusvolt": 3160,
			 "etouser_tod": 28, "etouser_tot": 28426, "etogrid_tod": 44, "etogrid_tot":
			 114409, "edischarge1_tod": 36, "edischarge1_tot": 58109, "eharge1_tod": 83,
			 "eharge1_tot": 61256, "elocalload_tod": 33, "elocalload_tot": 117140}}

egguy, 2023-02-28:
Thx for the report.

I think see the origin of the problem. I use the divider found in the layout configuration of grott. Currently, it marked as to divide per 100. 

```json
"SOC"                : {"value" :742, "length" : 2, "type" : "num", "divide" : 100}, 
```
I will do a hotfix of the mapping and probably open a ticket.

FlyAndy123, 2023-02-28:
Another weird mapping issue I have just found - it looks like 'pactogridr' is the actual export to the grid in Watts (W).
It seems to have been mapped to Energy Export (Today) in Wh
    "pactogridr": {
        "name": "Energy export (Today)",
        "device_class": "energy",
        "state_class": "measurement",
        "unit_of_measurement": "Wh",
        "state_class": "total",
        "icon": "mdi:solar-power",
I've confirmed this by comparing the values returned in Grott with the values shown on server.growatt.com

egguy, 2023-02-28:
Thank you for the information. 

I've also compared to https://github.com/muppet3000/homeassistant-growatt_server_api/blob/dev/custom_components/growatt_server_api/sensor_types/mix.py#L163 and in the manual it seems to be in W

[image]

I will to the change promptly

Meanwhile, do you have an idea of the meaning of S and T ? e.g. Pactogrid R

FlyAndy123, 2023-02-28:
I'm pretty sure that RST are the representing 3 phases.
RST stands for Reference, Secondary and Tertiary

In this case, the SPH5000 is only a single phase inverter, so only the R (or reference) would have any value.

egguy, 2023-02-28:
Can you try the new version of the add-on ?

closed 2023-09-21 by github-actions[bot]