egguy/grott-home-assistant-add-on #17: How to configure this?

Qonstrukt, 2023-02-28:
There's no documentation on the values that need to be put in the `mqtt` field.
I've made my ShineLink-X connect to this add-on, and I can see it's working, but no extra devices or entities are showing up under my MQTT integration in HA.
I'm already using MQTT for a couple of other things, so I know it's working.

egguy, 2023-02-28:
Yes, at the moment there is no configuration for a custom MQTT server. I will try to add this soon.

Qonstrukt, 2023-02-28:
No, not currently. I've already have it running in a separate container now, but I had to do some searching branches to find the correct latest version of your HA extension, which is really nice. 🙂 
Maybe a good idea to at least add some basic documentation on what it can do and can't to prevent more questions. And a link to your latest HA extension script, because the work is greatly appreciated!

DominikWrobel, 2023-02-28:
I've installed your addonn today, I have two growatt inverters. But how do I get the data to mqtt? No new entities show up. I have MQTT configured already in HA

egguy, 2023-02-28:
Hi, you need to follow the grott tutorial on how to reroute the traffic to the HA IP: https://github.com/johanmeijer/grott/wiki/Rerouting-Growatt-Wifi-TCPIP-data-via-your-Grott-Server

egguy, 2023-03-02:
@Qonstrukt I found a good way to add this functionality, I will probably do this next week. I don't have a computer at the moment. 

Do you have any suggestions of what I can add to the documentation?

egguy, 2023-03-07:
Hi @Qonstrukt,

I've updated the add-on to allow configuring a custom MQTT broker.

closed 2023-03-07 by egguy

Qonstrukt, 2023-03-07:
Great job, also on the expanded documentation!

egguy, 2023-03-07:
All thanks To zigbee2mqtt from where i stole a bunch of code :pirate_flag: