egguy/grott-home-assistant-add-on #18: Missing sensors on SPH3600

jakobschou, 2023-02-28:
Hi,

I have installed the "current" version (as I am guessing it is the bleeding edge being rc7 of grott_ha.py), and hoped that would have fixed my problem of missing sensor values/entities in Home Assistant MQTT auto discovery.

I can see the data being captured and converted to text in the log under MQTT jsonmsg: 
"eacharge_total": 994
"edischarge1_tot": 1219
"eharge1_tot": 1191

and repacked to ha_grott:
		 -  edischarge1_tod      :  7.4
		 -  edischarge1_tot      :  121.9
		 -  eharge1_tod          :  10.7
		 -  eharge1_tot          :  119.1

However when looking in my MQTT broker (mosquitto), I am not getting the entities. Some have values, others does not.

I have a SPH3600, and under invtype/inverter type I have put: sph
It does not work better with inverter type: default

Does anybody know why this is?

jakobschou, 2023-02-28:
My log (I have changed serials to X), if it has to be mapped, I will gladly help:

	 - Possible Inverter serial X
	 - Inverter serial not recognised - using inverter type default
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNX
	 - Grott data record date/time processing started
	 - date-time:  2023-02-28T20:23:45
	 - Grott values retrieved:
		 -  datalogserial        :  X
		 -  pvserial             :  X
		 -  recortype1           :  0
		 -  recortype2           :  124
		 -  pvstatus             :  6
		 -  pvpowerin            :  0.0
		 -  pv1voltage           :  0.0
		 -  pv1current           :  0.0
		 -  pv1watt              :  0.0
		 -  pv2voltage           :  0.0
		 -  pv2current           :  0.0
		 -  pv2watt              :  0.0
		 -  pvpowerout           :  0.0
		 -  pvfrequentie         :  0.0
		 -  pvgridvoltage        :  0.0
		 -  pvgridcurrent        :  0.0
		 -  pvgridpower          :  0.0
		 -  pvgridvoltage2       :  0.0
		 -  pvgridcurrent2       :  0.0
		 -  pvgridpower2         :  0.0
		 -  pvgridvoltage3       :  0.0
		 -  pvgridcurrent3       :  0.0
		 -  pvgridpower3         :  91082433.0
		 -  totworktime          :  0.0
		 -  pvenergytoday        :  0.0
		 -  pvenergytotal        :  0.0
		 -  epvtotal             :  16.2
		 -  epv1today            :  185.0
		 -  epv1total            :  293708.6
		 -  epv2today            :  9.3
		 -  epv2total            :  69.4
		 -  pvtemperature        :  36.0
		 -  pvipmtemperature     :  51.7
		 -  pbusvolt             :  428.0
		 -  nbusvolt             :  431.5
	 - MQTT jsonmsg: 
			 {"device": "X", "time": "2023-02-28T20:23:45", "buffered": "no",
			 "values": {"datalogserial": "X", "pvserial": "X",
			 "recortype1": 0, "recortype2": 124, "pvstatus": 6, "pvpowerin": 0,
			 "pv1voltage": 0, "pv1current": 0, "pv1watt": 0, "pv2voltage": 0,
			 "pv2current": 0, "pv2watt": 0, "pvpowerout": 0, "pvfrequentie": 0,
			 "pvgridvoltage": 0, "pvgridcurrent": 0, "pvgridpower": 0, "pvgridvoltage2":
			 0, "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0,
			 "pvgridcurrent3": 0, "pvgridpower3": 910824330, "totworktime": 0,
			 "pvenergytoday": 0, "pvenergytotal": 0, "epvtotal": 162, "epv1today": 1850,
			 "epv1total": 2937086, "epv2today": 93, "epv2total": 694, "pvtemperature":
			 360, "pvipmtemperature": 517, "pbusvolt": 4280, "nbusvolt": 4315}}

jakobschou, 2023-02-28:
With 'sph' inverter type I am getting all the data, however none are available in HA.

MQTT jsonmsg: 
			 {"device": "X", "time": "2023-02-28T21:07:16", "buffered": "no",
			 "values": {"datalogserial": "X", "pvserial": "X",
			 "recortype1": 0, "recortype2": 124, "pvstatus": 6, "pvpowerin": 0,
			 "pv1voltage": 0, "pv1current": 0, "pv1watt": 0, "pv2voltage": 0,
			 "pv2current": 0, "pv2watt": 0, "pvpowerout": 5695, "pvfrequentie": 4997,
			 "pvgridvoltage": 2333, "pvgridcurrent": 27, "pvgridpower": 5724,
			 "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0,
			 "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime":
			 2942306, "eactoday": 176, "pvenergytoday": 176, "eactotal": 1864,
			 "epvtotal": 1180, "epv1today": 93, "epv1total": 694, "epv2today": 62,
			 "epv2total": 481, "pvtemperature": 445, "pvipmtemperature": 426,
			 "pvboosttemp": 389, "bat_dsp": 520, "pbusvolt": 4280, "#nbusvolt": 4317,
			 "#ipf": 0, "#realoppercent": 0, "#opfullwatt": 0, "#deratingmode": 0,
			 "eacharge_today": 61, "eacharge_total": 994, "batterytype": 1,
			 "uwsysworkmode": 6, "systemfaultword0": 0, "systemfaultword1": 0,
			 "systemfaultword2": 0, "systemfaultword3": 0, "systemfaultword4": 32,
			 "systemfaultword5": 0, "systemfaultword6": 0, "systemfaultword7": 0,
			 "pdischarge1": 6600, "p1charge1": 0, "vbat": 519, "SOC": 40, "pactouserr":
			 0, "#pactousers": 0, "#pactousert": 0, "pactousertot": 0, "pactogridr": 0,
			 "#pactogrids": 0, "#pactogridt": 0, "pactogridtot": 0, "plocaloadr": 6600,
			 "#plocaloads": 0, "#plocaloadt": 0, "plocaloadtot": 6600, "#ipm": 0,
			 "#battemp": 24, "spdspstatus": 6, "spbusvolt": 2893, "etouser_tod": 98,
			 "etouser_tot": 3217, "etogrid_tod": 24, "etogrid_tot": 60,
			 "edischarge1_tod": 97, "edischarge1_tot": 1242, "eharge1_tod": 107,
			 "eharge1_tot": 1191, "elocalload_tod": 218, "elocalload_tot": 4388}}

egguy, 2023-03-01:
Hi,

Are you on your own MQTT server or the one in HA?

On the first data message received, do you have any `Pushing XXX configurations payload to HA` ? This will indicate that sensors are created in HA.

For the repacking, It's mostly HA doing the heavy lifting of the transformation with value templates, I don't really touch the values, excepted to add a last push sensor.

If correctly configured, you should have a `homeassistant/sensor/grott/XXXXX_sensor` config for each sensor.

When data is pushed, it should arrive at `homeassistant/grott/XXXX/state`

jakobschou, 2023-03-04:
Hi,

I am using the Mosquitto add-on through HA.

I have now figured out to get the MQTT data parsed manually via MQTT sensors in my config.

I found a topic/issue inside grott master branch with new MQTT setup for HA, and just cherrypicked the ones I needed which I could see were generated in the logs.

Funny part is that when using the "default" inverter type, the ha_grott will send data through MQTT auto-discovery, but not all data is included. But when using the "sph" inverter type, auto-discovery does not work, but I can get all the data manually, as explained before.

jakobschou, 2023-03-07:
Hi,

I am getting this error message:

Grott HA 0.0.7-RC8 - creating X config in HA, 85 to push
	Pushing 86 configurations payload to HA
	 - Grott extension processing error: ValueError('Publish topic cannot contain wildcards.')
	 - Traceback (most recent call last):
  File "/app/grottdata.py", line 677, in procdata
    ext_result = module.grottext(conf,result_string,jsonmsg)
  File "/app/grott_ha.py", line 942, in grottext
    publish_multiple(conf, configs_payloads)
  File "/app/grott_ha.py", line 852, in publish_multiple
    return multiple(msgs, **conf)
  File "/usr/lib/python3.10/site-packages/paho/mqtt/publish.py", line 177, in multiple
    client.loop_forever()
  File "/usr/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/lib/python3.10/site-packages/paho/mqtt/client.py", line 3029, in _packet_handle
    return self._handle_pubackcomp("PUBACK")
  File "/usr/lib/python3.10/site-packages/paho/mqtt/client.py", line 3529, in _handle_pubackcomp
    rc = self._do_on_publish(mid)
  File "/usr/lib/python3.10/site-packages/paho/mqtt/client.py", line 3489, in _do_on_publish
    on_publish(self, self._userdata, mid)
  File "/usr/lib/python3.10/site-packages/paho/mqtt/publish.py", line 68, in _on_publish
    _do_publish(client)
  File "/usr/lib/python3.10/site-packages/paho/mqtt/publish.py", line 40, in _do_publish
    client.publish(**message)
  File "/usr/lib/python3.10/site-packages/paho/mqtt/client.py", line 1233, in publish
    raise ValueError('Publish topic cannot contain wildcards.')
ValueError: Publish topic cannot contain wildcards.

egguy, 2023-03-07:
Ho nice one. Never seen this one before.

Did you activate the parameter `gincludeall` in the configuration? If it's the case, can you disable it for the moment?

jakobschou, 2023-03-07:
I just tried to disable 'gincludeall' and that actually fixed it! 

Weird - but thank you!

No I am getting the sensor data through MQTT auto discovery, but with some possible extra sensors missing, and some obsolete. But it works!

egguy, 2023-03-07:
Normally grott push only some measures, the measures prefixed by a `#` are not pushed (duplicates? Unknown mapping ?).

With the gincludeall flag, this function is disabled and everything is pushed, event prefixed measures.

In MQTT the `#` is a wildcard character, which is an illegal character for a topic. The topic names are: `$inverterID_jsonkey`, so with this parameter, it tries to create invalid topic.

This is why I need to add a protection in the future to prevent this problem.

closed 2023-03-07 by egguy