egguy/grott-home-assistant-add-on #20: Error on homeassistant with the latest grott-home-assistant-add-on

xhemp, 2023-04-09:
Greetings,

Been seeing some errors on my logs after the latest HA update (which could also be related to HA and not directly to your add-on):

[Logs here](https://paste.debian.net/1276852/)

If this doesn't belog here or you need more details on this or more verbosity, please, let me know.

xhemp, 2023-04-19:
Disabled the add-on for now and error is gone.

egguy, 2023-04-19:
I will take a look.

Did you change your inverter type at one point ?

xhemp, 2023-04-19:
Hey 👋🏻 
I have not, no, it's still the same (SPA3000).

xhemp, 2023-05-09:
Not sure if this has anything to do with the issue, I get it every time I start/restart HA:

2023-05-09 11:52:06.124 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grott/INV_SERIAL_70%_invpoweradjust/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/docs/mqtt/discovery/#discovery-topic
2023-05-09 11:52:06.136 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grott/INV_SERIAL_acchargeenergytotal /config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/docs/mqtt/discovery/#discovery-topic
2023-05-09 11:52:06.139 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grott/INV_SERIAL_battemp /config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/docs/mqtt/discovery/#discovery-topic
2023-05-09 11:52:06.182 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grott/INV_SERIAL_pactogrid t/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/docs/mqtt/discovery/#discovery-topic
2023-05-09 11:52:06.184 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grott/INV_SERIAL_pactogrids /config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/docs/mqtt/discovery/#discovery-topic

xhemp, 2023-06-15:
hey @egguy , just wondering where I could look into the TypeError: unsupported operand type(s) for /: 'float' and 'NoneType' error further? 
I combed everywhere but can't find where the problem lies seeing I can't see anything wrong on the json going to MQTT.

closed 2023-06-17 by xhemp

egguy, 2023-06-18:
Hi, I will try to take a look.
That's weird, but it seems to happen on some installations. I will try to do a bit of data collection on the tickets to see if there's a pattern.