egguy/grott-home-assistant-add-on #21: Missing proper install documentation

Petorrr, 2023-04-10:
Also I am experiencing the Growatt server API lockouts, accounts disables temporary. So I was looking to Grott and I successfully followed the installation procedure for this add-on for HA as described. Also changed the IP address of one of my data loggers to log to my local Home Assistant instance on which Grott is running after this install. The data is apparently sent through to the Growatt servers, I can see the updates coming in from that data logger on the growatt web server, so the proxy mode works of Grott. But how do I get the sensor data values into my Mosquito MQTT and HA, what do I need to do? I don’t see any new sensor appearing in my HA entities or sensors list….., it seems lacking in the documentation? Do I need to edit grott.ini, or add something in the configurations.yaml? A little more instructions may help here!

Petorrr, 2023-04-11:
Thanks so much for your comments. Apparently this morning it magically came up as device in Mosquitto broker, but not after I did some plugin updates (incl Mosquitto) yesterday evening. Perhaps the restart of HA triggered it this morning to work suddenly. Some snippet of the log entries shows it gets all data;

- Grott data record date/time processing started
	 - no or no valid time/date found, grott server time will be used (buffer records not sent!)
	 - Grott values retrieved:
		 -  datalogserial        :  XXXXXX
		 -  pvserial             :  YYYYYYY
		 -  pvstatus             :  1
		 -  pvpowerin            :  322.6
		 -  pv1voltage           :  143.7
		 -  pv1current           :  2.3
		 -  pv1watt              :  322.6
		 -  pv2voltage           :  0.0
		 -  pv2current           :  0.0
		 -  pv2watt              :  0.0
		 -  pvpowerout           :  316.2
		 -  pvfrequentie         :  49.9
		 -  pvgridvoltage        :  233.1
		 -  pvgridcurrent        :  1.3
		 -  pvgridpower          :  317.8
		 -  pvgridvoltage2       :  0.0
		 -  pvgridcurrent2       :  0.0
		 -  pvgridpower2         :  0.0
		 -  pvgridvoltage3       :  0.0
		 -  pvgridcurrent3       :  0.0
		 -  pvgridpower3         :  0.0
		 -  totworktime          :  459.2
		 -  pvenergytoday        :  8.2
		 -  pvenergytotal        :  148.0
		 -  epvtotal             :  150.5
		 -  epv1today            :  8.3
		 -  epv1total            :  150.5
		 -  epv2today            :  0.0
		 -  epv2total            :  0.0
		 -  pvtemperature        :  41.6
		 -  pvipmtemperature     :  45.7

However as device entities I only see a few sensors; Datalogger Serial, PV input (actual), PV1 Voltage, Serial and State. Could it be the rest is conflicting with the original data points from the Growatt Server which I also still run in parallel? They seem to have the same names……

egguy, 2023-04-11:
Also, If the logger has been disconnected for a long time, the logger start by pushing archived values, these values are not pushed in HA. This can delay the display of the first sensor by quite some time.

Depending on the inverters, if they are all the same, you can use the invertertype parameter to configure it. If it's a mix'n'match, you need to use the invertermap. You need to put the serial of the device with the type.

There's a link in the documentation of this repo on how to configure multiple inverter: https://github.com/johanmeijer/grott/wiki/Multi-Inverter-support, only the `If you have different types of inverters and need different layouts to process the data` part is revelant.

Petorrr, 2023-04-12:
Hi @egguy, just wanted to let you know everything is up and running now with two dataloggers (I use two similar Growatt inverters 1500 TL-X and 750 TL-X). All works perfect! I used the default settings of the configuration, and it captures nicely all data into HA, Grott also relaying everything still to the Growatt servers. So all good, thank you for the great work (and of course the Grott developer).

closed 2023-06-23 by egguy