egguy/grott-home-assistant-add-on #25: Update Interval

MrKohli, 2023-04-29:
Hey,
I've installed the add on about a month ago and it has been running ever since. 

One problem I'm having though that I only get new values every 5 minutes. I have set the update time interval of the data logger to 1 minute and with the official Grott docker container I do get the values every minute, but not with the add on.

My question is if I have to configure something something inside the add on to get the updates faster or if I'm missing something else.

Any help would be greatly appreciated.
Thanks a lot

egguy, 2023-04-29:
Hi,

This is strange. There is no modification to grott and the push depends on the data received by grott. You have a last push sensor to see when the last push by the extension has been done.

MrKohli, 2023-04-29:
Ok I just did that and I am getting this message 
[image]
every minute, but the Home Assistant sensors still only updates every 5 min. The last data push sensor also shows that the data is coming in every 5min.

MrKohli, 2023-05-03:
Ok I figured out what it is.
I had PVOutput activated and it only pushed the data there every 5min. Grott doesn't continue running the code if the data hasn't been pushed, so it never reaches the Home Assistant extension. I disabled PVOutput for now, and now I'm getting the data as soon as the inverter sends it.

closed 2023-05-03 by MrKohli