egguy/grott-home-assistant-add-on #35: Grott scanning interval

GroenZoom, 2023-07-26:
With your work i have got grott->mqtt running on HA2023.7.3 RPI3B+. Didn't succeed before with container image s that caused MQTT refusals. So far thank you for your efforts and work.

I had hoped that shortening the shinelink data interval from 5 down to 1 min that data resolution would improve.  PV_out_actual in Watts is OK and refreshing every minute but generated_energy_total_today and generatied_energy_total_lifetime in KWh still coming in only every 8~13 minutes.
Am i hoping for too much?

chesterflaps, 2023-08-10:
Did you manage to get to the bottom of this?

Also, how/where did you shorten the shinelink data interval?

GroenZoom, 2023-08-12:
You can change the push interval from default 5 min to 1 min in the Shinelan web UI login/password admin/admin.However resolution for power in Watts is 0,1 W and for Energy in kWh 0,1 kWh.Both are sensor entities that will update only when value changed n.b. more than 0,1 W and 0,1 kWh. The latter means that on our basic consumption of 200 ~ 300 Wh the entity will be updated only each 20 or 30 minutes. Power is updated each minute.RegardsMartin

closed 2023-09-19 by github-actions[bot]