egguy/grott-home-assistant-add-on #36: Amount of solar generated today is only reading from pv1

DanielB0nd, 2023-07-31:
I had a look at the MQTT data to try to confirm this and these are my findings,
[image]
This is the energy generated by pv1

This is what the supposibly combined generation is for pv1 and pv2
[image]

And this is what pv2 generated
[image]
```
{"datalogserial": "*******", "pvserial": "*******", "pvstatus": 6, "pvpowerin": 0, "pv1voltage": 108, "pv1current": 0, "pv1watt": 1, "pv2voltage": 0, "pv2current": 0, "pv2watt": 0, "pvpowerout": 401, "pvfrequentie": 5015, "pvgridvoltage": 2351, "pvgridcurrent": 9, "pvgridpower": 427, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime": 9529278, "eactoday": 53, "pvenergytoday": 53, "eactotal": 11140, "epvtotal": 12372, "epv1today": 53, "epv1total": 6841, "epv2today": 28, "epv2total": 5512, "pvtemperature": 322, "pvipmtemperature": 307, "pvboosttemp": 299, "bat_dsp": 531, "eacharge_today": 1, "eacharge_total": 75, "batterytype": 1, "uwsysworkmode": 6, "systemfaultword0": 0, "systemfaultword1": 0, "systemfaultword2": 0, "systemfaultword3": 0, "systemfaultword4": 32, "systemfaultword5": 0, "systemfaultword6": 0, "systemfaultword7": 0, "pdischarge1": 1200, "p1charge1": 0, "vbat": 531, "SOC": 94, "pactouserr": 0, "pactousertot": 0, "pactogridr": 246, "pactogridtot": 246, "plocaloadr": 1000, "plocaloadtot": 1000, "spdspstatus": 6, "spbusvolt": 3009, "etouser_tod": 29, "etouser_tot": 953, "etogrid_tod": 12, "etogrid_tot": 7987, "edischarge1_tod": 34, "edischarge1_tot": 2148, "eharge1_tod": 48, "eharge1_tot": 1906, "elocalload_tod": 84, "elocalload_tot": 5580, "grott_last_push": "2023-07-31T21:01:38.714977+00:00"}
``` 

As seen in the data taken from MQTT epv1today is 53 which is converted to 5.3kWh by the sensors added also stored in mqtt. However there is also pvenergytoday which matches pv1 at 53. 
epv2today is 28 (2.8kWh) so pvenergytoday should in theory be 81 (8.1kWh)

I dont know if this is an issue with GROTT as a whole or just the HA addon, so if I am misunderstanding something, or this is better suited to be posted on the GROTT page let me know and I will make an appropriate post on that page.

I am using the following version:
Home Assistant 2023.7.3
Supervisor 2023.07.1
Operating System 10.3
Grott edge edge-5353307094
Growatt Shine LAN-X
Home assistant is also running as a VM (incase this matters)

chesterflaps, 2023-08-10:
Sorry, not answering, just asking you a question.

How did you get the New Solar Generated today to work in the first place?

When Growatt worked, it had a total power generating and total power generated for the day, across both inverters. This was automatically there

DanielB0nd, 2023-08-10:
Heres a write up on how to achieve the custom sensors,

1. Install an addon to edit config files and yaml, personally I like Studio Code Server addon for this.
2. Add this line to you configuration.yaml `template: !include_dir_merge_list templates/`
3. Make a new folder called `templates`
4. In the new folder make a file called for example power.yaml the bit before .yaml doesnt matter as long as it has no spaces and .yaml is on the end (also caps matter I think)
5. Inside the new file add this https://pastebin.com/Y1GWiW5k (code blocks didnt work no multiple lines so pastebin it is)
6. Once you have pasted this in edit the sensor.inverter1 to your inverter generated or to pv1 and repeat for sensor.inverter2 to your second inverted generated or pv2

If you have any other questions about this lmk and I can try to walk you through it!

chesterflaps, 2023-08-10:
Thank you, I will take a look later.

Shame it is pointless right now because my Grott has gone to sh*t in the last 5 minutes.

there was a HA update so I updated and now when I start the Grott add on, it stops after a couple of seconds.

chesterflaps, 2023-08-10:
> 

I did this and tried to restart HA but got the below message

The system cannot restart because the configuration is not valid: Invalid config for [shell_command]: value should be a string for dictionary value @ data['shell_command']['template']. Got [{'sensor': [{'name': 'power_solar_today', 'unique_id': 'power_solar_today', 'state': "{{ states('sensor.be04070045_pv_output_actual') |int(0) + states('sensor.be04070099_pv_output_actual') |int(0) }}", 'unit_of_measurement': 'kWh', 'device_class': 'energy', 'state_class': 'total_increasing'}]}]. (See /config/configuration.yaml, line 23).

here is my code

```
- sensor:
    - name: power_solar_today
      #friendly_name: "Solar Generation - Today"
      unique_id: "power_solar_today"
      state: "{{ states('sensor.be04070045_pv_output_actual') |int(0) + states('sensor.be04070099_pv_output_actual') |int(0) }}"
      unit_of_measurement: "kWh"
      device_class: energy
      state_class: total_increasing
```

DanielB0nd, 2023-08-10:
The issue is that the sensors names and unique ids are the same, forgot to mention that my bad, this should work https://pastebin.com/ERCQWzWS

In future so you know, you dont specifically need unique ids, it just means that you can edit the sensor settings within the UI, not the yaml file (you dont need to make them match the sensors name, I do this so that there will never be the issue of one unique id matching another, it just makes it easier in my mind) and the sensor name is what you would call in home assistant dashboards e.g. power_solar_actual_today, and this can be anything I just prefix anything to do with power with power_ so that I know at a glance what its related to. (sorry for wall of text)

DanielB0nd, 2023-08-10:
Hang on, ignore that pastebin link I missed something that will cause it to error. Use this one https://pastebin.com/4wnuTquX

chesterflaps, 2023-08-10:
Worked that part out. I accidentally indented the code in the config file so HA thought it was part of the code above.

DanielB0nd, 2023-08-10:
There was also a ) out of place which I noticed which is why I sent the two pastebins. And do you mean you changed the sensor names and unique ids already?

chesterflaps, 2023-08-10:
[image]

chesterflaps, 2023-08-10:
[image]

DanielB0nd, 2023-08-10:
If you see on the state_class line of the first sensor there is a : on the end, if you remove that it should work (thanks for the screenshot)

chesterflaps, 2023-08-10:
> There was also a ) out of place which I noticed which is why I sent the two pastebins. And do you mean you changed the sensor names and unique ids already?

No I meant I managed to get it to restart

DanielB0nd, 2023-08-10:
any luck? that should have fixed it

chesterflaps, 2023-08-10:
> any luck? that should have fixed it

It stopped the error but I had to also remove the second word 'sensor' for it to stop giving an error

Like attached. The thing is, where do I find this sensor within HA to see if it is working? It seems that one of my inverters is not givng me any information since I did this but that could be down to slow feed from Growatt.
[image]

DanielB0nd, 2023-08-10:
If you go to Settings - Devices & Services - Entities (at the top) - and in the search type power_solar and there should be the two sensors, you do need to restart HA before they show up

chesterflaps, 2023-08-10:
Was it correct to me the second instance of the word 'sensor'?

chesterflaps, 2023-08-10:
> If you go to Settings - Devices & Services - Entities (at the top) - and in the search type power_solar and there should be the two sensors, you do need to restart HA before they show up

I thought it would be there but it hasn't shown up as yet. Trying another restart

chesterflaps, 2023-08-10:
Sadly not

[image]

DanielB0nd, 2023-08-10:
In developer tools there is a states, try searching in that maybe?

chesterflaps, 2023-08-10:
> In developer tools there is a states, try searching in that maybe?

Sadly not

DanielB0nd, 2023-08-10:
and you have added that line to the configuration.yaml?

chesterflaps, 2023-08-10:
I wonder, could it be because there should be 2 sensors?

Like this 

[image]

chesterflaps, 2023-08-10:
> and you have added that line to the configuration.yaml?

yes

DanielB0nd, 2023-08-10:
no it should be fine

chesterflaps, 2023-08-10:
[image]

chesterflaps, 2023-08-10:
Let me put a line in between and check

chesterflaps, 2023-08-10:
It is there now but see below

also, just getting unknowns across the board but I'm hoping that's due to the feed from growatt.

[image]

DanielB0nd, 2023-08-10:
and the sensors in the yaml are correct?

DanielB0nd, 2023-08-10:
what data are you getting from grott? and are you running grott edge with the built in intergration?

chesterflaps, 2023-08-10:
> and the sensors in the yaml are correct?

Just checking but there is only one showing anyway

[image]

DanielB0nd, 2023-08-10:
did you add the extra -sensor: if so try removing that

chesterflaps, 2023-08-10:
I notice your code is different.

you have 4 different entities. 2 different ones on the top part of the code and 2 totally different ones on the bottom part. Is you old one called old in one entity and pv1 in another?

I removed the second 'sensor'

DanielB0nd, 2023-08-10:
It should work then providing the sensors match up, only reason it wouldn't is that grott has no data on the sensors that you are using...

DanielB0nd, 2023-08-10:
I am now back, what does your grott config/console look like and are those sensors that are being read actually have data?

chesterflaps, 2023-08-10:
Sorry, had to go out.

I have put some images below to show. I ended up just doing one sensor to see if it would work and it did but I cannot work out the figures as it says 0kwh in the top right but when I press on the line graph, it says things like '117'

[image]
[image]
[image]
[image]

chesterflaps, 2023-08-10:
Actually, I guess the number is the power in watts at that time.

I now need to create another one to show the total generated across both inverters in the day. Seems the way I had it, the second one wouldn't work, almost like HA couldn't see it because it was thinking it was part of the first sensor and that's why neither worked.

This is very frustrating but I really appreciate your help through this.

chesterflaps, 2023-08-11:
created another sensor file called power2.yaml and all is fine.

Now I am trying to work out how I sue this sensor I have created in an automation. Any ideas?

DanielB0nd, 2023-08-11:
I use it on a power flow plus card (hacs)
[image]

chesterflaps, 2023-08-11:
> I use it on a power flow plus card (hacs) [image]

bajaysus

Next project ahoy!

Can I ask how you get your data for exporting to the grid? I am using Bright/Hildebrand at the moment but it only updates once a day with my smart meter usage and even though my smart meter tells me when I am exporting, I have no way of getting that information into HA.

DanielB0nd, 2023-08-11:
I get mine through a MyEnergy Zappi (EV charger), it reads solar and grid to only charge on green energy, it provides me with, grid (I can also get this through growatt because of the battery), all data relating to old solar. You could use something like a Shelly EM, but I dont know how easy that would be, another solution would be to use Hildebrand IHD but that is about £20 more, that was something I looked at as the zappi is cloud pull and there is no local intergration so often the house consumption which I calculate using solar, grid import/export, battery charge/discharge goes to below 0 but 0 is the min on the sensor

DanielB0nd, 2023-08-11:
And this isnt related to this post but you mention you are new to HA, one of the best things to do is backups, there are google backups and onedrive backups (with encryption to keep your data private)

chesterflaps, 2023-08-11:
Thanks for that.

did you manage to get PV1 and 2 on the total for the day? I am having the same issue. 

this is the code I used for it

```
 sensor:
    - name: power_solar_total_today
      unique_id: "power_solar_total_today"
      state: "{{ states('sensor.be04070***_generated_energy_today') |int(0) + states('sensor.be04070***_generated_energy_today') |int(0) }}"
      unit_of_measurement: "kWh"
      device_class: energy
      state_class: total_increasing
```

DanielB0nd, 2023-08-12:
this would work, https://pastebin.com/2Xnxq8qU, using the PV1 Generated and PV2 generated from two inverters

closed 2023-09-19 by github-actions[bot]