egguy/grott-home-assistant-add-on #37: Changes in Home Assistant 2024.2.0 Requires new Device-Names

HaareistNase, 2023-08-03:
Its seems like HA will Change some naming rules which affect Grott.
But Im to dump to understand what I can do in this case. 
Or do you have to change something in your wonderfull working addon?

[image]

chesterflaps, 2023-08-08:
I've literally just followed the steps to get this up and running and after reboot, I get the same message

egguy, 2023-08-09:
Hi,
It created a fix for grott_ha. I will publish soon a new add-on.

chesterflaps, 2023-08-10:
> Hi, It created a fix for grott_ha. I will publish soon a new add-on.

Is this what has caused my Grott to stop working?

there was an update on HA so I installed it and now Grott stops as soon as I start it

Maes152, 2023-08-14:
Has this issue been resolved? I saw an update come through for the addon but can't find any recent code changes that would imply it has been fixed.
Holding out on updating home assistant until this is fixed.

egguy, 2023-08-14:
Yes, this contains the updated grott ha add-on. I'm currently updating the repository to have a more proper release process.

closed 2023-09-21 by github-actions[bot]