egguy/grott-home-assistant-add-on #39: How do I add second inverter?

chesterflaps, 2023-08-08:
Hi all.

I have somehow managed to scrape through and get it to work but only for one of my two inverters.

Anyone know how I can add the second?

In layman's terms please as I am very new to this.

I see there is a link to a page saying what to do but I cannot make head nor tail of it and the files it makes reference to are not in my config

egguy, 2023-08-09:
Hi,
You just need to point the second logger to the same IP as the first. It should appear.

chesterflaps, 2023-08-10:
I swear I already did that but I have done it again and it works. 

Thank you.

Is there a way to get HA to show both loggers in one? When Growatt worked, it had a total power generating and total power generated for the day, across both inverters.

closed 2023-08-10 by egguy