egguy/grott-home-assistant-add-on #46: Addon is not starting - dependency dig missing

JannickWeichert, 2023-08-15:
Hello,

the addon stopped working today, probably after yesterdays update. HA is up-to-date and I removed and reinstalled the addon without an effect. Looks like the container doesn't include "dig". Could you check if there is something missing in the dependencies? 

Log output: 
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------
 Add-on: Grott edge - Home Assistant native MQTT integration - current
 The Growatt inverter monitor with current HA plugin integrated
-----------------------------------------------------------
 Add-on version: dev
 You are running the latest version of this add-on.
 System: Debian GNU/Linux 11 (bullseye)  (amd64 / qemux86-64)
 Home Assistant Core: 2023.8.2
 Home Assistant Supervisor: 2023.08.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[10:20:53] INFO: Preparing to start...
./script.sh: line 241: dig: command not found
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

daath, 2023-08-15:
I clicked "Rebuild" under add-ons, then I could start it again ...

JannickWeichert, 2023-08-15:
With default config, everything works fine. But when enabling `grott_mqtt: true`, the issue comes back.

egguy, 2023-10-05:
The latest version should correct this bug. Can you confirm ?

JannickWeichert, 2023-10-05:
Hi @egguy,

works now, thank you! 

Best Regards
Jannick

closed 2023-10-05 by JannickWeichert