egguy/grott-home-assistant-add-on #51: Growatt SPF 5000 ES Missing Data

julianjosias, 2023-09-28:
Hi Legends,

I have this all setup however, there is missing data. I have set spf in inverter type on the Grott ini.
Forgive me for my lack of knowledge as I am a total noob.
[image]
[image]
[image]
[image]
[image]

egguy, 2023-10-12:
Hi, how did you configure the SPH parameter? Did you use the interface? or created a file?

julianjosias, 2023-11-02:
Hi egguy. I used the interface grott ini spf (not sph).

julianjosias, 2023-12-12:
Just bumping this egguy.

irpyc, 2024-01-25:
Hi,
I also installed this addon, I´m so thankful for it, the terrible Growatt app is one of the main reason why I begun my HA journey huhu.
Thanks a lot for all the work.

But I also have some missing value for my system (SPF 5000, with Pylontech battery).
The most annoying ones are the fact I don´t have a "Battery charge from Solar today".
The only information about charge is the "Battery discharge power" showing negative value.

Therefor I cannot integrate it in the HA "Energy" Tab :(
I already spent hours trying to find a workaround...

Also I didn´t found values differentiating Power generated from Power in Bypass mode, but I´m not that far in my configuration.

Some screenshot (not that much sun today, and the inverter is still in Bypass mode, since it start at 30%)

[image]

[image]

[image]

[image]