egguy/grott-home-assistant-add-on #54: Wrong calculation

GeraldPape, 2023-09-30:
[image]

Summing up PV1 and PV2 gives 22,4
Generated energy shows only 14,7
[image]

Why?

Btw. Shine cloud shows correct value, pvoutput also received wrong summation

egguy, 2023-10-05:
I need to take a look, but the addon+extension is only relaying the information decrypted by grott. This can be either a bad decoding or the values, or a bad report from the inverter.

As a workaround, you can use the `settings -> devices -> helper` and create a sum of the two sensors to have the correct value.

martijntonies, 2023-10-10:
Could be a record decoding error like I think I have?
https://github.com/johanmeijer/grott/issues/443

GeraldPape, 2023-10-13:
Checked on the Growatt Cloud - Values are summing up correctly:
[image]

so seems to be some decoding issue?

egguy, 2023-11-20:
I will find if there is a decoding error. Perhaps it’s an invalid register.
Meanwhile, you can use a home assistant helper to generate the sum of the 2 sensors. 

You can use this likg to go to: [Settings > Devices & Services > Helpers](https://my.home-assistant.io/redirect/helpers) and click the add button at the bottom right. There is a "combine sensors" and then you can do a sum

closed 2024-01-30 by egguy