egguy/grott-home-assistant-add-on #56: Day production always start with Minus Solar?

martin22cph, 2023-10-03:
Everyday, my PV or Solar always start with minus produktion?

See picture...

[image]

martin22cph, 2023-10-04:
Again today it did the same? 

[image]

martin22cph, 2023-10-05:
This was a error 40, myswelf. The minus starting the day, was the prior days production, as i need to select the right sensor in Home Assistent. Selecting Lifetime solar energy, fixed the daily starting minus. :)

egguy, 2023-10-05:
Yes, the energy calculation in HA works with a constantly incrementing value. I will try to update the documentation.

If you want to contribute to the documentation, I would be glad to accept your contribution.