egguy/grott-home-assistant-add-on #57: output to pvoutput

Vinisz, 2023-10-04:
Hi,

I know normal grott is able to output to "pvoutput" , how should I configure this for the addon ?

TjappieW, 2023-10-05:
I edited the file grott.ini with Studio Code Server. The file was empty and i copy/pasted the pvoutput settings (from a grott.ini example in the grott repo) in to the file and edit them with my apikey and systemid. 

```
[PVOutput]
# PVOutput parameters definitions
pvoutput = True
apikey = yourapikey 
# Data upload limit (in minutes)
pvuplimit = 5
# Use this if you have one inverter
systemid = yoursystemid
```

Vinisz, 2023-10-05:
Thanks , I found out that you can just add it in the JAML as well.. however, maybe something else is needed.. where is the grott.ini located ?

TjappieW, 2023-10-05:
I found the file in  the folder "config/grott".

corove, 2023-10-10:
> I edited the file grott.ini with Studio Code Server. The file was empty and i copy/pasted the pvoutput settings (from a grott.ini example in the grott repo) in to the file and edit them with my [...]

Hi.
I am fairly new to Grott and Growatt, and struggles somehow to set up this integration fully. I have got i t to send data to the Grott integration, but there is no mqtt sensors populating or showing up. Maybe i have to put in this code aswell? But where do I locate my APIkey and SystemID?

egguy, 2023-10-11:
Hello,

This add-on is mainly to have a local version of grott. This will forward the data to growatt and send a copy to your local HA. It don't support connecting to growatt server to get the data.

You probably want to use the native grott HA integration, or use @muppet3000 (https://github.com/muppet3000/) HACS integration.

corove, 2023-10-11:
> Hello,
>
> This add-on is mainly to have a local version of grott. This will forward the data to growatt and send a copy to your local HA. It don't support connecting to growatt server to get the [...]

Hi.

yes i understand that this is the local version, and it is this i want. The muppet3000 integration stopped working and will nor reconnect to growatt servers, so therefore i am trying this metode instead. Before flashing my shinewifi end running esphome on it.😉 but i don’t know where to locate the API keys and systemID mentioned in the code above?

muppet3000, 2023-10-11:
@corove - I support a different integration for working with Grott, this one: https://github.com/muppet3000/homeassistant-grott
All you need to do to make it work with the @egguy add-on is to install it and then tweak the installation of the @egguy  add-on slightly:
- Check `enable grott MQTT`
- Disable `HA integration`

Then the MQTT devices should be marked as disabled, and the HACS integration will receive the message instead.

You don't need to do anything with the `[PVOutput]` section of the config, that's for if you want to publish your data to https://pvoutput.org/

egguy, 2023-10-27:
@corove for pvoutput key, I think you can find the information over here: https://pvoutput.org/help/index.html

closed 2023-10-27 by egguy