egguy/grott-home-assistant-add-on #59: Update from 0.5 to 0.6

tatadile, 2023-10-05:
The command '/bin/ash -o pipefail -c apk add --update --no-cache python3=3.11.4-r0 && ln -sf python3 /usr/bin/python && python3 -m ensurepip' returned a non-zero code: 1

misterburma, 2023-10-06:
Hi,
First, thank you for the addon :-)

I'm receiving also the same error when trying to do the update. Hereby my logs.

Home Assistant 2023.10.0
Supervisor 2023.10.0

```
23-10-06 09:23:54 ERROR (MainThread) [supervisor.docker.addon] Can't build egguy/addon-amd64-grott-ha-addon:bd5c208: The command '/bin/ash -o pipefail -c apk add --update --no-cache python3=3.11.4-r0 && ln -sf python3 /usr/bin/python && python3 -m ensurepip' returned a non-zero code: 1
23-10-06 09:23:54 ERROR (MainThread) [supervisor.docker.addon] Build log: 
Step 1/19 : ARG BUILD_FROM=ghcr.io/hassio-addons/base:14.1.0

Step 2/19 : FROM $BUILD_FROM

 ---> 04f18ed3e104

Step 3/19 : ENV LANG C.UTF-8

 ---> Using cache

 ---> 5fbb7815f712

Step 4/19 : ENV PYTHONUNBUFFERED=1

 ---> Using cache

 ---> dc1d0f8920c2

Step 5/19 : RUN apk add --update --no-cache python3=3.11.4-r0 && ln -sf python3 /usr/bin/python && python3 -m ensurepip

 ---> Running in 538f84f08a82

fetch https://dl-cdn.alpinelinux.org/alpine/v3.18/main/x86_64/APKINDEX.tar.gz

fetch https://dl-cdn.alpinelinux.org/alpine/v3.18/community/x86_64/APKINDEX.tar.gz

ERROR: unable to select packages:

  python3-3.11.6-r0:
    breaks: world[python3=3.11.4-r0]

    satisfies: python3-3.11.6-r0[python3~3.11]

Removing intermediate container 538f84f08a82
```

tatadile, 2023-10-11:
> The command '/bin/ash -o pipefail -c apk add --update --no-cache python3=3.11.4-r0 && ln -sf python3 /usr/bin/python && python3 -m ensurepip' returned a non-zero code: 1

i just tried it now again and it worked thanks anyway

BTW
GREAT add-on!!!!!!!!!!!!!!!!

egguy, 2023-10-12:
Your welcome.

closed 2023-10-12 by egguy