egguy/grott-home-assistant-add-on #60: Not seeing export and load consumption, only PV generation info

doenau, 2023-10-06:
I just switched over to the Grott add-on due to the HA Growatt integration no longer working and this being the recommended approach.
I've got both Grott (b6f2f24) and Grott - HA Extension (2.7.8) installed - noting that it won't let me update to bd5c208.

This is the result when running the HA Extension, so for the moment, I have given up on that option and it is turned off.
[image]

So, running Grott b6f2f24, I don't seem to get any EXPORT values, which I did with the Growatt integration.
I have the ShineStick connecting back via the ShineLAN box.

Is this normal, and this add-on is just lacking the export detail? Or have I done something wrong?

[image]

Neflardio, 2023-10-06:
Missing the inverter type configuration might be the issue.
[image]

If that's not the issue, posting the the log  when running in verbose mode is a good starting point.

doenau, 2023-10-07:
Thanks @Neflardio - I have tried the inverter type options and they give me worse results, either nothing, or a bunch of invalid values in all the wrong places.
anyway - for now I will close this as a duplicate, and follow issue #47 that I just found, which seems much the same.
https://github.com/egguy/grott-home-assistant-add-on/issues/47

closed 2023-10-07 by doenau