egguy/grott-home-assistant-add-on #63: Unable to install grott-stable after uninstall

Geo-Ron, 2023-10-11:
After a uninstall, I am unable to reinstall the addon.

[image]

Supervisor log:
```
23-10-11 20:45:49 ERROR (MainThread) [supervisor.docker.addon] Invalid build environment, can't build this add-on!
23-10-11 20:46:41 ERROR (MainThread) [supervisor.docker.addon] Invalid build environment, can't build this add-on!
23-10-11 20:48:38 ERROR (MainThread) [supervisor.docker.addon] Invalid build environment, can't build this add-on!
23-10-11 20:50:19 ERROR (MainThread) [supervisor.docker.addon] Invalid build environment, can't build this add-on!
```

HomeAssistant:
```
Home Assistant 2023.10.1
Supervisor 2023.10.0
Operating System 10.5
Frontend-versie: 20231005.0 - latest
```

egguy, 2023-10-11:
Hi,
I think I see the bug, this comes from a packaging error. It should be corrected soon.

egguy, 2023-10-12:
I think the problem should have been solved. Could you confirm?

Geo-Ron, 2023-10-12:
I can not confirm.
Error message is the same.
As is the change number on the top right (044729b) I think.

I will check again this evening.

Geo-Ron, 2023-10-12:
[image]

Verifiëren! I can confirm it is ok!
Thanks

closed 2023-10-12 by Geo-Ron