egguy/grott-home-assistant-add-on #64: SPF 3000TL LVM-ES not sending data

waite902, 2023-10-17:
My  data logger seems to be connecting to my grott server but never passes data beyond what I suspect is the ACK/NACK stuff.  I assume there is supposed to be a handshake of some sort before the actual telemetry data is sent that isn't happening

System Details
Home Assistant 2023.10.1 running on a raspberry pi 4 Home assistant operating system 10.5
Growatt SPF 3000TL LVM-ES with Shine Wifi-S
grott-home-assistant-add-on installed via HACS

Shine wifi-s  setup with the raspberry pi's IP address, 192.168.0.50, default port (5279),  update interval set to 1 min

Grott config
-inverter type:  spf
gmode: proxy
Enable grott MQTT:  Enabled
Home Assistant plugin:  disabled

in /config/grott/grott.ini, I added "minrecl = 1" for debugging purposes

It has been running several days.  Below is a representative snip from the log file.  Any clues?

- layout   :  T060119SPF
	 - no matching record layout found, try generic
	 - Record layout used :  T060119SPF
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00010006003501194444443041343831384e00000000000000000000000000000000000000000
		 010001132343a41313a36303a32433a35353a38463fe8
	 - Grott data ack record or data record not defined no processing done
	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.5', 37702), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x00\x01\x00\x06\x00\x30\x01\x19\x03\x36\x2b\x47\x20\x40\x4c\x76\x4a\x21\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x66\x61\x78\x45\x7e\x40\x41\x46\x57\x4c\x5a\x77\x5c\x5a\x47\x78\x73
	 - Grott automatic protocol detection
	 - Grott data record length 56
	 - layout   :  T060119SPF
	 - no matching record layout found, try generic
	 - Record layout used :  T060119SPF
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00010006003001194444443041343831384e00000000000000000000000000000000000000000
		 011000c3139322e3136382e302e35301907
	 - Grott data ack record or data record not defined no processing done
	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.5', 37702), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x00\x01\x00\x06\x00\x28\x01\x19\x03\x36\x2b\x47\x20\x40\x4c\x76\x4a\x21\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x65\x61\x70\x41\x75\x45\x56\x2c\x5c
	 - Grott automatic protocol detection
	 - Grott data record length 48
	 - layout   :  T060119SPF
	 - no matching record layout found, try generic
	 - Record layout used :  T060119SPF
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00010006002801194444443041343831384e00000000000000000000000000000000000000000
		 0120004353237395b3d
	 - Grott data ack record or data record not defined no processing done
	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.5', 37702), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x00\x01\x00\x06\x00\x24\x01\x19\x03\x36\x2b\x47\x20\x40\x4c\x76\x4a\x21\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x64\x61\x74\x95\xef
	 - Grott automatic protocol detection
	 - Grott data record length 44
	 - layout   :  T060119SPF
	 - no matching record layout found, try generic
	 - Record layout used :  T060119SPF
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00010006002401194444443041343831384e00000000000000000000000000000000000000000
		 0130000e1a8
	 - Grott data ack record or data record not defined no processing done
	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.5', 37702), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x00\x01\x00\x06\x00\x38\x01\x19\x03\x36\x2b\x47\x20\x40\x4c\x76\x4a\x21\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x63\x61\x60\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c
		 \x1f\x2a\x37\x2f\x39\x11\xaf
	 - Grott automatic protocol detection
	 - Grott data record length 64
	 - layout   :  T060119SPF
	 - no matching record layout found, try generic
	 - Record layout used :  T060119SPF
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00010006003801194444443041343831384e00000000000000000000000000000000000000000
		 0140014585858585858585858585858585858585858585865db
	 - Grott data ack record or data record not defined no processing done
	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.5', 37702), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x00\x01\x00\x06\x00\x2b\x01\x19\x03\x36\x2b\x47\x20\x40\x4c\x76\x4a\x21\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x62\x61\x73\x47\x69\x42\x41\x47\x4f\x44\xd1\x3e
	 - Grott automatic protocol detection
	 - Grott data record length 51
	 - layout   :  T060119SPF
	 - no matching record layout found, try generic
	 - Record layout used :  T060119SPF
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00010006002b01194444443041343831384e00000000000000000000000000000000000000000
		 0150007332e302e302e30a579
	 - Grott data ack record or data record not defined no processing done

egguy, 2023-10-17:
The first data often appears in the next few minutes. You should see quickly some data coming.

Do you run the addon in verbose mode?

waite902, 2023-10-17:
Yes, it is in verbose mode.

I do get data every few minutes, but it is always a short record without the telemetry data.  Here is the last entry coming from the shinewifi-s.  It is typical (maybe identical) of what is received.

	 - Growatt packet received:
		  <socket.socket fd=7, family=2, type=1, proto=0, laddr=('172.30.33.5', 5279), raddr=('192.168.0.101', 14403)>
	 - Growatt original Data:
		 \x00\x01\x00\x06\x00\x24\x01\x19\x03\x36\x2b\x47\x20\x40\x4c\x76\x4a\x21\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x68\x61\x6b\x5e\x6e
	 - Grott automatic protocol detection
	 - Grott data record length 44
	 - layout   :  T060119SPF
	 - no matching record layout found, try generic
	 - Record layout used :  T060119SPF
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00010006002401194444443041343831384e00000000000000000000000000000000000000000
		 01f001f2a29
	 - Grott data ack record or data record not defined no processing done

egguy, 2023-10-27:
This is weird. do you use the growatt server? I've seen some install using different server.

waite902, 2023-10-27:
Yes.  Previously I was pointed at the us growatt server (server-us.growatt.com) if that makes a difference

waite902, 2023-10-27:
To check, I changed the growatt server in the config to the us server and seem to get the same results

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=7, family=2, type=1, proto=0, laddr=('172.30.33.5', 5279), raddr=('192.168.0.209', 17024)>
	 - Data less then minimum record length, data not processed
	 - ('47.251.35.20', 5279) has disconnected

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=7, family=2, type=1, proto=0, laddr=('172.30.33.5', 5279), raddr=('192.168.0.209', 17024)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=7, family=2, type=1, proto=0, laddr=('172.30.33.5', 5279), raddr=('192.168.0.209', 17024)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 51802), raddr=('47.251.35.20', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=7, family=2, type=1, proto=0, laddr=('172.30.33.5', 5279), raddr=('192.168.0.209', 17024)>
	 - Data less then minimum record length, data not processed

egguy, 2023-11-05:
Did you make the change in the .ini or in the interface? By default, grott send the info to `server.growatt.com (47.91.67.66)`, you need to change the IP. (this accepts a DNS)

closed 2023-12-13 by github-actions[bot]