egguy/grott-home-assistant-add-on #67: Growatt 6000TL3-XH, can't receive packages in the Add-On

DR-Apples, 2023-10-30:
Hello everyone, 

I tried to connect my Growatt 6000TL3-XH to my Home Assistant via the ShineLan-X stick. I assigned a static IP to the inverter and visited it to change the server settings to my HA IP (192.168.1.2). 
[image]

After that, I installed the stable Grott branch (2.7) and entered the following setting based on my inverter: Growatt 6000TL3-XH
[image]

When starting the Add-On, I can't receive any packages/information and I get the following Logs: (password blurred)
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: Grott stable branch (2.7)
 The Growatt inverter monitor with current HA plugin integrated
-----------------------------------------------------------
 Add-on version: 0.1.5
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.1  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2023.10.5
 Home Assistant Supervisor: 2023.10.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[20:44:11] INFO: Preparing to start...
	 **********************************************************************************
	 - Grott - libscrc not installed, no CRC checking only record validation on length!
	 **********************************************************************************
Grott Growatt logging monitor : 2.7.8

Grott process configuration file

Grott process environmental variables

	Pulled 'ginvtype=tl3' from the environment

	Pulled 'gnomqtt=False' from the environment

	Pulled 'gmqttip=172.30.33.0' from the environment

	Pulled 'gmqttport=1883' from the environment

	Pulled 'gmqttauth=True' from the environment

	Pulled 'gmqttuser=addons' from the environment

	Pulled 'gmqttpassword=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX' from the environment

	Pulled 'gextension=True' from the environment

	Pulled 'gextname=grott_ha' from the environment

	Pulled 'gextvar={"ha_mqtt_host": "core-mosquitto", "ha_mqtt_port": "1883", "ha_mqtt_user": "addons", "ha_mqtt_password": "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX", "ha_mqtt_retain": False}' from the environment

Grott override settings if set in commandline

Grott process json layout files

Grott proxy mode started
	 **********************************************************************************
	 - Grott - libscrc not installed, no CRC checking only record validation on length!
	 **********************************************************************************
Hostname : 8ce4abae-grott
IP :  172.30.33.4 , port :  5279 
```

I am running the mosquito broker Add-On on the same Raspberry Pi and want to connect it to this when it works.

I hope someone knows the solution to the problem, Thank you in advance @egguy !

egguy, 2023-11-05:
Did you reboot the logger after the change?

As for the add-on, if you have indicated the LAN IP of HA, you should see new lines appear in the logs. The new lines won't show up automatically, you need to click on refresh.

DR-Apples, 2023-11-16:
Hi Egguy,

After some trail and error, did i manage to fix the problem. It turns out that by changing the any of the information in de configuration tab, the grott add-on stops updating the data. 

Add-on is now working correct for me with the default config.

But there could be a bug in the config page if you adjust the inverter type to the one i should need (?, don't have that much knowledge about it so i could be wrong)

santiacho-jt, 2024-07-30:
Hey! 
I updated the datalogger IP as suggested, but the settings were restored after a while. I'm not sure why and who, but I can't update it. Is anyone experiencing this problem? 
Steps:
- Resolve Domain: OFF
- Server IP: (HA internal IP) CONN OK
- Data transfer interval: 1 (this is the only thing that is being saved)

But changes for the Resolve domain and Server IP get restored by default 🤔 😞 

Rebooted the ShineLan-X stick the datalogger, and updated to the 7.0.0.9 version.