egguy/grott-home-assistant-add-on #69: Wrong values maybe SP2000 isn't supported?

giangy11, 2023-11-16:
Hi,
Is the growatt sp2000 (storage inverter) with shine wifi s supported? I have a lot of wrong values
[image]

giangy11, 2023-11-17:
I found another user added locally the below lines to add compatibility with Growatt model sp-2000 (storage inverter)
Is it possible to add this lines to the grottconf.py file?

self.recorddict3 = {"T06NNNN": {
"decrypt" : {"value" :"True"},
"datalogserial" : {"value" :16, "length" : 10, "type" : "text","incl" : "yes"},
"pvserial" : {"value" :76, "length" : 10, "type" : "text"},
"date" : {"value" :136, "divide" : 10},
"recortype1" : {"value" :150, "length" : 2, "type" : "num","incl" : "no"},
"recortype2" : {"value" :154, "length" : 2, "type" : "num","incl" : "no"},
"Generation_status" : {"value" :158, "length" : 2, "type" : "num"},
"pvpowerin" : {"value" :162, "length" : 4, "type" : "num", "divide" : 10,"incl" : "yes"},
"Solar_voltage" : {"value" :170, "length" : 2, "type" : "num", "divide" : 10,"incl" : "yes"},
"Solar_current" : {"value" :174, "length" : 2, "type" : "num", "divide" : 10,"incl" : "yes"},
"Solar Generation" : {"value" :178, "length" : 4, "type" : "num", "divide" : 10,"incl" : "yes"},
"bat_Volt" : {"value" :206, "length" : 2, "type" : "num", "divide" : 10,"incl" : "yes"},
"Battery_Charge" : {"value" :218, "length" : 2, "type" : "num", "divide" : 10,"incl" : "yes"},
"Grid_AC_Voltage" : {"value" :226, "length" : 2, "type" : "num", "divide" : 10,"incl" : "yes"},
"Import_From_Grid" : {"value" :238, "length" : 2, "type" : "num", "divide" : 10,"incl" : "yes"},
"Temperature" : {"value" :242, "length" : 2, "type" : "num", "divide" : 10,"incl" : "yes"},
"export_to_Grid" : {"value" :250, "length" : 2, "type" : "num", "divide" : 10,"incl" : "yes"},
"test_Bat_dischg" : {"value" :346, "length" : 2, "type" : "num", "divide" : 10,"incl" : "yes"},
"Solar_today" : {"value" :358, "length" : 4, "type" : "num", "divide" : 10,"incl" : "yes"},
"Solar_total" : {"value" :366, "length" : 4, "type" : "num", "divide" : 10,"incl" : "yes"}

egguy, 2023-11-23:
The container give access to the full configuration on the /config/grott. In therory, you cam create a T06NNNN.json file in this folder and it will be loaded

closed 2023-12-31 by github-actions[bot]