egguy/grott-home-assistant-add-on #73: grott-home-assistant-add-on log file location

LightMoon, 2023-12-14:
Hi @egguy!

I am trying to troubleshoot the missing PV3 sensor issue and need to know where the extension log file is stored and how it can be retrieved.

#https://github.com/muppet3000/homeassistant-grott/issues/6#issuecomment-1855117828

egguy, 2023-12-14:
Hi,
There is not really a log file. You can enable the verbose mode in the addon and go see the logs in  the log panel or in the system.

From here you can download them.

LightMoon, 2023-12-14:
Thanks egguy, unfortunately when verbose is enabled lots of information will get overwritten on every refresh. May I know where the location of system logs is?

egguy, 2023-12-14:
Grott doesn't implement a of log file. Everything is printed on the console.
If some logs are collected by home assistant, this would be available throught home assistant.

closed 2024-01-21 by github-actions[bot]