egguy/grott-home-assistant-add-on #74: PV3 is missing on TL3 configuration - Grott Add-on

LightMoon, 2023-12-18:
Hi Etienne!

I need some assistance in understanding why PV3 data isn't being picked up for TL3 inverter. [muppet3000](https://github.com/muppet3000)
[Chris Straffon](https://github.com/muppet3000) checked out the logs and ruled out the issue on the component side. 

More about the history of the issue and logs are available here. Feel free to let me know if you need more info.
#https://github.com/muppet3000/homeassistant-grott/issues/6#issuecomment-1858571029

egguy, 2024-01-17:
Hi,

is it related to the `invtype=min` configuration ?

LightMoon, 2024-01-18:
> Hi,
> 
> Is it related to the `invtype=min` configuration ?

Yes, it is. I initially started exploring Grott by setting the inverter type as TL3, but later on realised my inverter was MIN. However, when I switched to MIN I lost anything related to the smart meter. I had to set the inverter type to default to get the data related to `SPM2DBE044` visible on HA. I already opened an issue on here, hopefully, someone can look into the logs and let me know what is happening.
#https://github.com/johanmeijer/grott/issues/508

* I am using muppet3000's integration for registering sensors and unfortunately, anything sensor related to **Today** Imported from the grid, Exported to the Grid, or Load Consumption has not been registered.

LightMoon, 2024-01-18:
#https://github.com/muppet3000/homeassistant-grott/issues/64
I did a bit of research that could be very related to this issue mentioned.