egguy/grott-home-assistant-add-on #76: Grott doesn't start with MQTT enabled

RienduPre, 2024-02-02:
My Grott add-on stopped working a couple of weeks ago, during my vacation

After some testing I have the following results:
With MQTT disabled I get the log below, with one error: ERROR: Got unexpected response from the API: Service not enabled
The addon starts and the data from my Growatt dongles are flowing in, but no MQTT messages, which is normal because MQTT is disabled in this case.
The data is also forwarded tot the growatt cloud servers

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: Grott stable branch (2.7)
 The Growatt inverter monitor with current HA plugin integrated
-----------------------------------------------------------
 Add-on version: 0.1.7
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.4  (aarch64 / odroid-n2)
 Home Assistant Core: 2024.2.0b2
 Home Assistant Supervisor: 2024.01.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[16:05:53] INFO: Preparing to start...
[16:05:59] ERROR: Got unexpected response from the API: Service not enabled
	 **********************************************************************************
	 - Grott - libscrc not installed, no CRC checking only record validation on length!
	 **********************************************************************************
Grott Growatt logging monitor : 2.7.8

Grott process configuration file

Grott process environmental variables

	Pulled 'ginvtype=default' from the environment

	Pulled 'ginvtypemap={"DNG0A4702R": "tl3"}' from the environment

	Pulled 'gnomqtt=True' from the environment

	Pulled 'gmqttip=xxx.xxx.xxx.xxx' from the environment

	Pulled 'gmqttport=1883' from the environment

	Pulled 'gmqttauth=true' from the environment

	Pulled 'gmqttuser=xxxxx' from the environment

	Pulled 'gmqttpassword=xxxxxx' from the environment

Grott override settings if set in commandline

Grott process json layout files

Grott proxy mode started
	 **********************************************************************************
	 - Grott - libscrc not installed, no CRC checking only record validation on length!
	 **********************************************************************************
Hostname : 8ce4abae-grott
IP :  172.30.33.8 , port :  5279 `

When I just enable MQTT, nothing works anymore
Same error in the log, but also **./script.sh: line 244: GROTT_CONFIG_MQTT_HOST: unbound variable**
My mosquitto addon is working without issue, I have other add-on using it without issues

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: Grott stable branch (2.7)
 The Growatt inverter monitor with current HA plugin integrated
-----------------------------------------------------------
 Add-on version: 0.1.7
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.4  (aarch64 / odroid-n2)
 Home Assistant Core: 2024.2.0b2
 Home Assistant Supervisor: 2024.01.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[16:15:59] INFO: Preparing to start...
[16:16:05] ERROR: Got unexpected response from the API: Service not enabled
./script.sh: line 244: GROTT_CONFIG_MQTT_HOST: unbound variable
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

I also tried the beta version of HA and your add-on, with exactly the same results.
Many thanks in advance for your help.

Best regards,
Rien

RienduPre, 2024-02-05:
Discard this issue. After restarting Mosquitto broker it started working again, very  weird.

closed 2024-02-05 by RienduPre

egguy, 2024-02-06:
It is weird. It seems there is a change with the new ha release. You're the second person reporting this.