egguy/grott-home-assistant-add-on #77: Growatt MOD 7000TL3-XH missing Battery Data

Jaybeste, 2024-02-02:
Hi Experts, 

am I missing something or is my inverter not fully supported? My battery is not visible. I tried setting my inverter type to default, as well as all SPH, SPF or tl3, but nothing really changed. 

[image]
[image]
[image]

Spielnichts, 2024-02-02:
i have the same issue.
i am using a mid-20ktl3-XH and get no battery data.

manuelachterberg, 2024-05-05:
Unfortunately I have the same issue with my MOD 4000TL3-XH (v3) on Grott Beta branch (2.8)
Current version: 0.1.7.
There are no battery values in the MQ:
{"device": "XGD6CLR32S", "time": "2024-05-05T14:51:11", "buffered": "no", "values": {"datalogserial": "XGD6CLR32S", "pvserial": "HHM1DCP0E6", "voltage_l1": 2384, "voltage_l2": 2373, "voltage_l3": 2381, "Current_l1": 4, "Current_l2": 59, "Current_l3": 8, "act_power_l1": -1092, "act_power_l2": 13933, "act_power_l3": -191, "app_power_l1": 1116, "app_power_l2": 14120, "app_power_l3": 1827, "react_power_l1": 118, "react_power_l2": -2012, "react_power_l3": -1810, "powerfactor_l1": -994, "powerfactor_l2": 989, "powerfactor_l3": -133, "pos_rev_act_power": 12650, "pos_act_power": 12650, "rev_act_power": 12650, "app_power": 19862, "react_power": -3793, "powerfactor": 957, "frequency": 499, "L1-2_voltage": 4121, "L2-3_voltage": 4118, "L3-1_voltage": 4127, "pos_act_energy": 27002, "rev_act_energy": 615}}