egguy/grott-home-assistant-add-on #84: suddenly stopped sending mqtt data via ha_plugin

f0m3, 2024-07-17:
Hi, 
the addon suddenly stopped to send mqtt data via the plugin. I didnt change anything intentionally.
there is just no mqtt update to mosquitto anymore.
I tried to reinstall, checked if the user "addons" exists and so on...
The proxy mode seems to work because the growatt server gets updated and only updated when the add-on is running.

Here is my config:
`data_path: /config/grott
gmode: proxy
grott_mqtt: false
ginvtype: default
ginvtypemap: "{\"SPE*******\":\"SPH\", \"FVJ*****\":\"default\"}"
retain: false
verbose: true
ha_plugin: true
mqtt: {}
gnomqtt: false`

and this is the log:
`-----------------------------------------------------------
 Add-on: Grott Beta branch (2.8)
 The Growatt inverter monitor with current HA plugin integrated
-----------------------------------------------------------
 Add-on version: 0.1.5
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.4  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.7.2
 Home Assistant Supervisor: 2024.06.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[10:26:12] INFO: Preparing to start...
`
[..]
`

T06NNNNXMIN  : 
{'decrypt': {'value': 'true'}, 'pvserial': {'value': 76, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 136, 'divide': 10}, 'group1start': {'value': 150, 'length': 2, 'type': 'num', 'incl': 'no'}, 'group1end': {'value': 154, 'length': 2, 'type': 'num', 'incl': 'no'}, 'pvstatus': {'value': 158, 'length': 2, 'type': 'num', 'divide': 1}, 'pvpowerin': {'value': 162, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 174, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 178, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 186, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 190, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 194, 'length': 4, 'type': 'num', 'divide': 10}, 'pv3voltage': {'value': 202, 'length': 2, 'type': 'num', 'divide': 10}, 'pv3current': {'value': 206, 'length': 2, 'type': 'num', 'divide': 10}, 'pv3watt': {'value': 210, 'length': 4, 'type': 'num', 'divide': 10}, 'pv4voltage': {'value': 218, 'length': 2, 'type': 'num', 'divide': 10}, 'pv4current': {'value': 222, 'length': 2, 'type': 'num', 'divide': 10}, 'pv4watt': {'value': 226, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 250, 'length': 4, 'type': 'num', 'divide': 10}, 'pvfrequentie': {'value': 258, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 262, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 266, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 270, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 278, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 282, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 286, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 294, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 298, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 302, 'length': 4, 'type': 'num', 'divide': 10}, 'vacrs': {'value': 310, 'length': 2, 'type': 'num', 'divide': 10}, 'vacst': {'value': 314, 'length': 2, 'type': 'num', 'divide': 10}, 'vactr': {'value': 318, 'length': 2, 'type': 'num', 'divide': 10}, 'ptousertotal': {'value': 322, 'length': 4, 'type': 'num', 'divide': 10}, 'ptogridtotal': {'value': 330, 'length': 4, 'type': 'num', 'divide': 10}, 'ptoloadtotal': {'value': 338, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 346, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvenergytoday': {'value': 354, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 362, 'length': 4, 'type': 'num', 'divide': 10}, 'epvtotal ': {'value': 370, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1today ': {'value': 378, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 386, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2today': {'value': 394, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2total': {'value': 402, 'length': 4, 'type': 'num', 'divide': 10}, 'epv3today': {'value': 410, 'length': 4, 'type': 'num', 'divide': 10}, 'epv3total': {'value': 418, 'length': 4, 'type': 'num', 'divide': 10}, 'etousertoday': {'value': 426, 'length': 4, 'type': 'num', 'divide': 10}, 'etousertotal': {'value': 434, 'length': 4, 'type': 'num', 'divide': 10}, 'etogridtoday': {'value': 442, 'length': 4, 'type': 'num', 'divide': 10}, 'etogridtotal': {'value': 450, 'length': 4, 'type': 'num', 'divide': 10}, 'eloadtoday': {'value': 458, 'length': 4, 'type': 'num', 'divide': 10}, 'eloadtotal': {'value': 466, 'length': 4, 'type': 'num', 'divide': 10}, 'deratingmode': {'value': 502, 'length': 2, 'type': 'num', 'divide': 1}, 'iso': {'value': 506, 'length': 2, 'type': 'num', 'divide': 1}, 'dcir': {'value': 510, 'length': 2, 'type': 'num', 'divide': 10}, 'dcis': {'value': 514, 'length': 2, 'type': 'num', 'divide': 10}, 'dcit': {'value': 518, 'length': 2, 'type': 'num', 'divide': 10}, 'gfci': {'value': 522, 'length': 4, 'type': 'num', 'divide': 1}, 'pvtemperature': {'value': 530, 'length': 2, 'type': 'num', 'divide': 10}, 'pvipmtemperature': {'value': 534, 'length': 2, 'type': 'num', 'divide': 10}, 'temp3': {'value': 538, 'length': 2, 'type': 'num', 'divide': 10}, 'temp4': {'value': 542, 'length': 2, 'type': 'num', 'divide': 10}, 'temp5': {'value': 546, 'length': 2, 'type': 'num', 'divide': 10}, 'pbusvoltage': {'value': 550, 'length': 2, 'type': 'num', 'divide': 10}, 'nbusvoltage': {'value': 554, 'length': 2, 'type': 'num', 'divide': 10}, 'ipf': {'value': 558, 'length': 2, 'type': 'num', 'divide': 1}, 'realoppercent': {'value': 562, 'length': 2, 'type': 'num', 'divide': 1}, 'opfullwatt': {'value': 566, 'length': 4, 'type': 'num', 'divide': 10}, 'standbyflag': {'value': 574, 'length': 2, 'type': 'num', 'divide': 1}, 'faultcode': {'value': 578, 'length': 2, 'type': 'num', 'divide': 1}, 'warningcode': {'value': 582, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword0': {'value': 586, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword1': {'value': 590, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword2': {'value': 594, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword3': {'value': 598, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword4': {'value': 602, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword5': {'value': 606, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword6': {'value': 610, 'length': 2, 'type': 'num', 'divide': 1}, 'systemfaultword7': {'value': 614, 'length': 2, 'type': 'num', 'divide': 1}, 'invstartdelaytime': {'value': 618, 'length': 2, 'type': 'num', 'divide': 1}, 'bdconoffstate': {'value': 630, 'length': 2, 'type': 'num', 'divide': 1}, 'drycontactstate': {'value': 634, 'length': 2, 'type': 'num', 'divide': 1}, 'group2start': {'value': 658, 'length': 2, 'type': 'num', 'incl': 'no'}, 'group2end': {'value': 662, 'length': 2, 'type': 'num', 'incl': 'no'}, 'edischrtoday': {'value': 666, 'length': 4, 'type': 'num', 'divide': 10}, 'edischrtotal': {'value': 674, 'length': 4, 'type': 'num', 'divide': 10}, 'echrtoday': {'value': 682, 'length': 4, 'type': 'num', 'divide': 10}, 'echrtotal': {'value': 690, 'length': 4, 'type': 'num', 'divide': 10}, 'eacchrtoday': {'value': 698, 'length': 4, 'type': 'num', 'divide': 10}, 'eacchrtotal': {'value': 706, 'length': 4, 'type': 'num', 'divide': 10}, 'priority': {'value': 742, 'length': 2, 'type': 'num', 'divide': 1}, 'epsfac': {'value': 746, 'length': 2, 'type': 'num', 'divide': 100}, 'epsvac1': {'value': 750, 'length': 2, 'type': 'num', 'divide': 10}, 'epsiac1': {'value': 754, 'length': 2, 'type': 'num', 'divide': [... cut ...]
	extvar:             	 {'ha_mqtt_host': 'core-mosquitto', 'ha_mqtt_port': '1883', 'ha_mqtt_user': 'addons', 'ha_mqtt_password': 'imooroGh2giijaiToo6moopuwaqu1ohsh7Thah9eiWeiNg9tuiMoo9yia7aey0Ai', 'ha_mqtt_retain': False}

Grott proxy mode started
	 **********************************************************************************
	 - Grott - libscrc not installed, no CRC checking only record validation on length!
	 **********************************************************************************
Hostname : 8ce4abae-grott-current
IP :  172.30.33.6 , port :  5279 `

egguy, 2024-07-17:
If you are getting data in growatt website but not grott it is highly possible you are affected by the growatt stick reset problem

Could you please try this : https://github.com/egguy/addon-grott-beta/issues/51#issuecomment-2231910621

rhpijnacker, 2024-07-17:
I had exactly this problem yesterday. My ShineLink-LAN stick had defaulted back to server.growatt.com and I had to reconfigure it to send the data to my local grott server.

f0m3, 2024-07-18:
Thank you so much. That ofc. solved the problem.
Thanks for the work - enjoy the coffee!

f0m3, 2024-07-18:
Not sure if this is interesting for you:
I resetted the wifisticks ip to my local grott. It worked immidately. But just for one dataset. It was automatically reset again to growatts server.
I tried it now with gblockcmd: true before i reset the shine stick..

Lets see what happens...

closed 2024-08-25 by github-actions[bot]