egguy/grott-home-assistant-add-on #87: Grott Retain Issues: Troubleshooting Status Unavailability After Home Assistant Restarts

LightMoon, 2024-08-27:
Hi!

I'm not sure if this is due to a misconfiguration or a genuine bug, but I've noticed that the Grott States become unavailable whenever Home Assistant restarts. The issue persists till receiving new message from the inverter.

```
data_path: /config/grott
gmode: proxy
grott_mqtt: true
ginvtype: min
ginvtypemap: ""
retain: true
verbose: true
ha_plugin: false
mqtt: {}
gnoipf: true
gblockcmd: true
gpvoutput: true
gpvsystemid: 103487
gpvapikey: *************************
gpvdisv1: false
gtime: server
gtimezone: Australia/Sydney
```

```
Grott settings:

_Generic:
	version:             	 2.8.3
	verbose:             	 True
	trace:               	 False
	config file:         	 /config/grott/grott.ini
	minrecl:             	 100
	decrypt:             	 True
	compat:              	 False
	invtype:             	 min
	invtypemap:          	 {}
	include_all:         	 False
	blockcmd:            	 True
	noipf:               	 True
	time:                	 server
	sendbuf:             	 True
	timezone:            	 Australia/Sydney
	valueoffset:         	 6
	offset:              	 6
	inverterid:          	 automatic
	mode:                	 proxy
	grottip              	 default
	grottport            	 5279
_MQTT:
	nomqtt               	 False
	mqttip:              	 172.30.33.0
	mqttport:            	 1883
	mqtttopic:           	 energy/growatt
	mqttmtopic:          	 False
	mqttmtopicname:      	 energy/meter
	mqttinverterintopic: 	 False
	mqtttretain:         	 False
	mqtttauth:           	 True
	mqttuser:            	 addons
	mqttpsw:             	 **secret**
_Growatt server:
	growattip:           	 server.growatt.com
	growattport:         	 5279
_PVOutput:
	pvoutput:            	 True
	pvdisv1:             	 False
	pvtemp:              	 False
	pvurl:               	 https://pvoutput.org/service/r2/addstatus.jsp
	pvapikey:            	 *********************
	pvinverters:         	 1
	pvsystemid:          	 103487
_Influxdb:
	influx:             	 False
	influx2:            	 False
	database:           	 grottdb
	ip:                 	 localhost
	port:               	 8086
	user:               	 grott
	password:           	 **secret**
	organization:       	 grottorg
	bucket:             	 grottdb
	token:              	 **secret**
_Extension:
	extension:          	 False
	extname:            	 grottext
	extvar:             	 {'none': 'none'}
```

typxxi, 2024-09-13:
that is the usual behaviour.
No message, no value.

During night without a hybridinverter you get errors in HA or unavailable unless you check those values.

LightMoon, 2024-09-20:
> that is the usual behaviour. No message, no value.
> 
> During night without a hybridinverter you get errors in HA or unavailable unless you check those values.

What is the purpose of `retain: true` ?

egguy, 2024-09-30:
Hi, the [retain relate to a flag for the MQTT broker.](https://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages/) .

The idea is to keep the last message stored in mosquitto. In case of reboot, this allows HA to keep the last state, or else all the sensor would be `unavaialbe` until grott start. There is a side effect, HA will consider this value is now the value of the semsor it time it checks. 

If your PV produce 200W when the stick stop sending new message, then HA will consider the inverter continu to produce this value as it is always returned, because of the retain.
You can look in the bug report of this repot. there is multiple report of people having inverter producing power at night because of this.

To prevent this, I added an expiration relative to the freshness of the messages (15 minute), after this, the sensor will be marked as unavailable and won't be considered into the calculation of the produced power.

closed 2024-11-07 by github-actions[bot]